stack-chan/stack-chan #27: Which capacitor is recommended?

Tiryoh, 2021-11-13:
Hi, I have a question about the capacitor.

According to the BOM in `schematics/README.md`, it says that 25V 100uF capacitor is recommended, 

https://github.com/meganetaaan/stack-chan/blob/b1527039dd3d9fd7924e1a690a36d9ebe434e341/schematics/README.md#L25

but the 10V 100uF capacitor is listed in the parts list in the same document. What are the requirements for the capacitor?

https://github.com/meganetaaan/stack-chan/blob/b1527039dd3d9fd7924e1a690a36d9ebe434e341/schematics/README.md#L50

There seem to be several types of the same series of capacitors sold by Sengoku Densho.

| Manufacturer Product Number | Rated Voltage | Capacitance | Diamiter (D) | Height (L) | URL |
| --- | --- | --- | --- | --- | --- |
| 1EUTES101M0 | 25V | 100uF | 6.3 mm | 11 mm | https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=46S8-5GL7 |
| 1CUTES101M0 | 16V | 100uF | 5.0 mm | 11 mm | https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=2678-5GLR |
| 1AUTES101M0 | 10V | 100uF | 5.0 mm | 11 mm | https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=76S8-53LK |

[image]
source: https://www.mepc.jp/store/pdf/UTES_Series.pdf

Considering the above, I assume that the requirements for capacitors are as follows, is that right?

* Rated Voltage > 10V
* Diamiter < 6.3 mm
* Height < 11 mm

meganetaaan, 2021-11-14:
Thank you for trying checking docs carefully!

>According to the BOM in schematics/README.md, it says that 25V 100uF capacitor is recommended,

My mistake. Since the voltage on the power supply line is 5V, so the rated 10V of capacitor is enough.

>Considering the above, I assume that the requirements for capacitors are as follows, is that right?

Looks good. I think the size of capacitor is also should be documented. I'll fix README.

meganetaaan, 2021-11-14:
Fixed! If you find another inconvenience about it, please reopen.

closed 2021-11-14 by meganetaaan

Tiryoh, 2021-11-14:
Looks great! Thanks!

PR #28 (2021-11-14): Fix/27/capacitor requirement
This PR fixes #27 

- Fixed reference URL
- Wrote size requirements of capacitor on README
commit a05a6b61 (2021-11-14): Fix #27 fix capacitor page on BOM as recommended rated voltage