stack-chan/stack-chan #29: Which type of servo should I use?

yoshipon, 2021-11-28:
Thank you for making and sharing the nice robot!! 

I'm now thinking about which type of servo motors should I use for making a stack-chan.
I would appriciate it if I could know the pros. and cons. of the PWM and TTL servos.

\# 素晴らしい活動ありがとうございます！

meganetaaan, 2021-12-10:
That's a good question! I've written up the advantages and disadvantages of each. It's about the SG90 and RS304MD that I'm currently using, but I think the former generally applies to PWM and the latter to serial servo (TTL).

### SG90

Pros: Very inexpensive (~500yen), easy to find, simple to control (just give PWM waveforms of varying lengths). Can be used with PortC (serial communication).
Cons: Difficult to change the angle smoothly. Inrush current is high and may cause the M5Stack to power down in rare cases (with the current board configuration); gets very hot and may emit smoke when overloaded, such as when the angle exceeds the physical limits of the stack-chan.

### RS304MD

Pros: Complex control such as speed limit is possible. Since it can read the current servo angle, it can perform functions such as changing the action according to the direction of Stuck Chan's face.
Cons: Relatively expensive (~3,000 yen), cannot be used with PortC. Because the servo is one size larger, it sticks out slightly behind the main unit.

yoshipon, 2021-12-29:
Thank you for your clear answers! Sorry for my late reply... I now finally got my holidays.
How about putting this information into `schematics/README.md`? I'll try to make a (draft) PR to discuss it in a few days.

meganetaaan, 2022-01-07:
Merged! These kind of question and maintenance really helps me making repository good for beginners. Feel free to post another question if you have!

closed 2022-01-07 by meganetaaan