stack-chan/stack-chan #30: Battery charging capability

yoshipon, 2021-11-28:
Sorry for my multiple issues, but I thought it's better to separate issues having different topics.

Does the stack-chan's board (or M5Stack itself) have a capability to charge the Li-Po battery?
Should I buy some charging device?
Since I could not read this info from the READMEs, please let me know about it.

If it's preferable, I'd like to make a PR adding  the information from your answer to the README.

Thanks,
Yoshiaki

meganetaaan, 2021-12-05:
Thank you for your comment!

>Does the stack-chan's board (or M5Stack itself) have a capability to charge the Li-Po battery?

Yes it does. Since the battery is connected to BAT pin of M5Stack. It can be charged through USB power of M5Stack.
It would be great if you could give me a PR.

yoshipon, 2021-12-29:
Thank you again! As in #29, I'll make a draft PR to discuss updating `./README.md`.

meganetaaan, 2022-01-07:
Merged! Thank you so much!

closed 2022-01-07 by meganetaaan