stack-chan/stack-chan #32: NOT Working with M5Stack Core2

SteveJos, 2021-12-04:
After flashing the latest master (dockerised procedure) like described in the [README](https://github.com/meganetaaan/stack-chan/blob/main/firmware/README.md) the M5Stack Core2 is not booting or showing anything on the screen. 

- [x] flashed master as well as the feature/core2 branch. 
- [x] did a clean erase und reflashed multiple times
- [x] the stock M5 firmware is still working
- [x] no compile/flash issues

the only thing that I did not is attaching any kinds or servos. The M5Stack Core2 is in default configuration.

Unfortunately there is almost no community instructions on how to customize (disable servos) the FW (at least english readable)

Any thoughts?

thx

meganetaaan, 2021-12-10:
This happens because my npm script description assumed M5Stack Basic.

```
    "deploy": "cd stackchan && mcconfig -m -p esp32/m5stack",
```

For CORE2, it should look like the following.

```
    "deploy:m5stack_core2": "cd stackchan && mcconfig -m -p esp32/m5stack_core2"
```

We will modify this command to include it in the npm script later.

[I'm also attaching a patch ](https://gist.github.com/meganetaaan/24c96edaad1a012e076b49e3bcea85ff) for the ModdableSDK code that needs to be modified in one place to make M5Stack CORE2 work. I've already sent a [pull request to upstream](https://github.com/Moddable-OpenSource/moddable/pull/745), so if everything is ok, it will be merged.

Appreciate your help! It still requires a lot of effort to maintain the firmware documentation compared to the hardware and circuitry.

Translated with www.DeepL.com/Translator (free version)

meganetaaan, 2021-12-12:
Fixed! If you find another inconvenience please reopen it.

closed 2021-12-12 by meganetaaan

SteveJos, 2021-12-21:
@meganetaaan thank you very much! It is working now!

commit 537068a8 (2021-12-12): Merge pull request #33 from meganetaaan/fix/32/startup_m5stack_core2

Fix #32 fix startup sequence of m5stack core2