stack-chan/stack-chan #40: Connections of SG90s

yoshipon, 2022-01-02:
Hi, now I'm testing the stack-chan's firmware with Core 2, and I, if I correctly understand the schematic, found that the "pan" servo is assigned to J3 (GPIO 27) and the "tilt" servo is assigned to J2 (GPIO 19).

This assignment may be confusing because of the following two reasons:
1. The "tilt" servo is located upper than the "pan" servo, whereas the J2 is lower than the J3.
2. The correct assignment is not documented in the READMEs.

I would like you to confirm my understanding, and, if you like, I'll make a PR to add information to some README.

By the way, I also found that `schematics/README.md` is a bit old because the images/photos are for the older version board, and we need to solder additional jump patterns (JP1, 2, 6, and 8).
If you also like, I'll try to take photos of my board and update the READMEs.

I imagine that it is so tough to maintain OSH by oneself, please feel free to ask me to make some help.

meganetaaan, 2022-01-02:
Thank you for your report!
I thought I had straightforwardly designed the circuit so that the upper pin (J3) would be the tilt axis and the lower pin (J2) would be the pan axis.
In fact, the wrong was on firmware side. I've noticed that [the pin assignments ](https://github.com/meganetaaan/stack-chan/blob/0f94cdfe8d3133fb8304153465551bc9b1c4255c/firmware/stackchan/manifest.json#L52)in the firmware were reversed.

```json
    "platforms": {
        "esp32/m5stack_core2": {
            "config": {
                "servo": {
                    "pwmPan": 27,
                    "pwmTilt": 19
                }
            }
        }
    }
```

I was able to solve this odd inversion by assigning GPIO 19 to pan and 27 to tilt as shown below.

```json
    "platforms": {
        "esp32/m5stack_core2": {
            "config": {
                "servo": {
                    "pwmPan": 19,
                    "pwmTilt": 27
                }
            }
        }
    }
```

The wrong motor connection due to the bug can be very dangerous, especially if the tilt axis tries to move beyond its physical limits. I'll merge this change ASAP!
It would be great if you could add some documentation on the cable assembly based on this pin assignment.

meganetaaan, 2022-01-07:
Should be fixed now. Thank you!

closed 2022-01-07 by meganetaaan