stack-chan/stack-chan #66: suggested improvements to manifest.json

phoddie, 2022-07-27:
I would like to suggest a few small improvements to [firmware/stackchan/manifest.json](https://github.com/meganetaaan/stack-chan/blob/main/firmware/stackchan/manifest.json).

1. For the HTTP module, use the manifest provided by the Moddable SDK. Simply add `$(MODDABLE)/modules/network/http/manifest.json` to the `include` section and remove http from the `modules` and `preload` sections. This will keep Stackchan in-sync with future changes to HTTP.
2. Add `button` and `sg90-driver` to the `preload` section. This saves memory.
3. Increase `creation/keys` from 128 to 512. This is the number of unique property names a mod can contain. In my experiments, 128 was not enough. Each key in this table uses 4 bytes, so this uses an additional 1.5 KB. The ESP32 has lots of free memory running Stackchan, so this shouldn't be a problem.

Additionally, the M5Stack Core 2 port now [implements support for the touch buttons](https://github.com/Moddable-OpenSource/moddable/commit/3223338b8f0dc6e96295f426e36630dff977ba3d). I think this means that you may be able to remove the `button` module and the [code to set-up the pseudo-buttons](https://github.com/meganetaaan/stack-chan/blob/0c52a56b08896ac949b15ca048b5c2cfa1c15af1/firmware/stackchan/main.ts#L28-L43) from main.ts.

Thank you for considering these suggestions.

meganetaaan, 2023-10-23:
Very late reply but these are now all resolved in the latest [`manifest.json`](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/stackchan/manifest.json) .
In addition, for reusability, [I separated the manifest file into modules](https://github.com/meganetaaan/stack-chan/pull/202).

Thank you for your suggestion!

phoddie, 2023-10-23:
Thank you. Just one FYI.

> Increase creation/keys from 128 to 512

This was the best practice when this issue was opened. Since that time, XS has been enhanced to automatically grow the keys array when necessary. Therefore, it is no longer necessary to preallocate the worst-case number of keys.

It is probably a good idea to keep this change in place because it will be a little faster than growing the key array and may reduce memory fragmentation.  The incremental amount of memory is 1.5 KB (4 bytes for each reserved key entry) which is probably not an issue on the devices that typically host Stack-chan.