stack-chan/stack-chan #95: Preference for Wi-Fi connection Wi-Fi接続設定の追加

meganetaaan, 2023-01-07:
**Is your feature request related to a problem? Please describe.**

So far there is a way to specify SSID and password in the `mcconfig` command line option or in the manifest file, but there is no way to specify it when writing the mod.

Wi-Fiの設定について、今の所mcconfigのコマンドラインオプションかmanifestファイルで指定する方法がありますが、modの書き込み時に指定する方法がありません。

**Describe the solution you'd like**

Add preference to set Wi-Fi SSID and password

Wi-FiのSSIDとパスワードを設定する手段を追加します。

stc1988, 2023-01-10:
[Node-RED MCU Edition](https://github.com/phoddie/node-red-mcu/blob/main/mods/readme.md) sets Wi-Fi setting from mcrun.
It helps us to implement this feature.

meganetaaan, 2023-10-23:
https://github.com/meganetaaan/stack-chan/blob/382ca8b3ce02334b50434b442849771d7abb163b/firmware/docs/setting-preferences-web_ja.md

ブラウザから設定を変更する機能を実装・公開しました。

closed 2023-10-23 by meganetaaan