stack-chan/stack-chan #104: modをアンインストールする

stc1988, 2023-01-09:
**既存の機能に問題がある場合、その説明**

modをインストール後にデフォルト(modのインストール前)の挙動に戻したいケースがあると思います。
そこで、modをアンインストールする方法が必要と考えています。

**あなたが望む解決策**。

過去にModdable SDK側の[ディカッション](https://github.com/Moddable-OpenSource/moddable/discussions/846)でmodのアンインストール方法を確認すると、`serial2xsbug`を使えるようです。
この手段でstack-chanのmodをアンインストールできることを確認しています。

ただし、serial pathを調べる必要があるため自動的に検出できるコマンド化されると望ましいです。

**検討した代替案**

現状stack-chanのmodの実装では、modに`onRobotCreated`が実装されていなければデフォルトの挙動となるため、このような空のmodをインストールすることでも実現できると思います。

s1tnk, 2023-11-08:
modのアンインストール機能、あるといいですね。
esptool.pyは自動でデバイスを探すので、その動作を模したスクリプトを書いてみました。

シリアルデバイスに順に接続を試みて、最初に接続に成功したデバイスのシリアルのパスを表示し、ステータス0で終了します。該当デバイスが無い場合はステータス1で終了します。

```
import os
import sys
from esptool import get_port_list, get_default_connected_device, DEFAULT_CONNECT_ATTEMPTS, ESPLoader

def blockPrint():
    sys.stdout = open(os.devnull, 'w')

def enablePrint():
    sys.stdout = sys.__stdout__

blockPrint()
ser_list = get_port_list()

esp = get_default_connected_device(
    ser_list,
    port=os.environ.get("ESPTOOL_PORT", None),
    connect_attempts=DEFAULT_CONNECT_ATTEMPTS,
    initial_baud=os.environ.get("ESPTOOL_BAUD", ESPLoader.ESP_ROM_BAUD)
)

if esp:
    enablePrint()
    print(esp.serial_port)
    exit(0)
else:
    exit(1)

```

stc1988, 2024-08-17:
#256 で代替案としてFlashを削除する方法を実装しました。
いったんはその方法をドキュメントに残して本件クローズし、mod領域のみ削除したいケースが多くなるようであれば再度検討します。

stc1988, 2024-08-17:
#256 がマージされたのでクローズします

closed 2024-08-17 by stc1988