stack-chan/stack-chan #115: Make a single binary ready to flash

meganetaaan, 2023-01-29:
**既存の機能に問題がある場合、その説明**

現在、ｽﾀｯｸﾁｬﾝのファームウェアは自分でModdableとesp-idfをインストールしてビルドする必要があります。[インストールスクリプトの導入](https://github.com/meganetaaan/stack-chan/pull/108)でだいぶ簡単にはなりましたが、初心者の方やとりあえず試してみたい人などには環境構築が障壁となります。

**あなたが望む解決策**。

ｽﾀｯｸﾁｬﾝのビルド時に、M5Stackにそのまま書き込める単一のバイナリを生成したいです。

M5StackはM5Burnerというファームウェアフラッシュ用のツールを[公開しています](https://docs.m5stack.com/en/download)M5Burnerにはユーザ製のアプリケーションを登録、公開できます。

ｽﾀｯｸﾁｬﾝもM5Burnerに対応したイメージを作り、登録できると良さそうです。

**検討した代替案**

WebSerialに対応した[esptool](https://adafruit.github.io/Adafruit_WebSerial_ESPTool/)を使うとウェブブラウザだけで書き込みができそうです。ｽﾀｯｸﾁｬﾝとしてはプロプライエタリなM5Burnerよりもオープンなツールを使いたいので、こちらも検討する価値はあります。

**その他**
機能リクエストに関するその他のコンテキストやスクリーンショットをここに追加してください。

stc1988, 2023-03-05:
[ESP Web Tools](https://github.com/esphome/esp-web-tools/)を試しました。

`manifest.json` 作成のために必要なファイルやオフセットの情報は、`$MODDABLE/build/tmp/<platform>/<device>/debug/<appication>/xsProj-esp32/build/flasher_args.json` で確認ができます。書き込むためのバイナリファイルはこの`flasher_args.json`に記載されたパスまたは、`$MODDABLE/build/bin/<platform>/<device>/debug/<appication>`に配置されています。

`de/v1.0`ブランチのソフトウェアが起動できることを確認できたので、Webサイト経由での配布ができそうです。

stc1988, 2023-04-02:
> 書き込むためのバイナリファイルはこのflasher_args.jsonに記載されたパスまたは、$MODDABLE/build/bin/<platform>/<device>/debug/<appication>に配置されています。

`mcbundle`を使うと上記バイナリファイルを別途指定したファイルと合わせてパッケージングができ、また、対象のデバイスも`manifest.json`に書くだけとなるので、少しだけリリースプロセスが楽になるかもしれません。

stc1988, 2023-04-13:
メモレベルですが、以下の影響を要確認
https://github.com/Moddable-OpenSource/moddable/discussions/1087

meganetaaan, 2023-06-28:
ESP Web Toolsを使ってM5Stack Basic向けのｽﾀｯｸﾁｬﾝが書き込めるページを試験的にデプロイしました。

https://meganetaaan.github.io/stack-chan/web/

stc1988, 2023-07-01:
おそらく`dev/1.0`のししかわ版のファームウェアなので、サーボの動作までは確認できていませんが、ページからの書き込みは確認できました

meganetaaan, 2023-10-23:
書き込み方法のドキュメントを追加して公開しました。

https://github.com/meganetaaan/stack-chan/blob/382ca8b3ce02334b50434b442849771d7abb163b/firmware/docs/flashing-firmware-web_ja.md

closed 2023-10-23 by meganetaaan