stack-chan/stack-chan #116: Failed to generate-apidoc

stc1988, 2023-01-29:
**不具合の概要**
`npm run generate-apidoc` を実行すると失敗します。

**環境**

* HostOS: macOS
* commit: https://github.com/meganetaaan/stack-chan/commit/ea40fd989cc3100fb3d9393a399945dcd53eff55

**その他**

Docker環境でないため、[tsconfig.json](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/tsconfig.json)内のパスを手元の環境に合わせて更新しましたが、一部モジュールの定義が見つからないようでした。

stackchanビルド時に生成された`tsconfig.json`に内容を差し替えることで成功しました。
`tsconfig.json`の内容が古いように見えます。

meganetaaan, 2023-01-29:
報告ありがとうございます！
今のtsconfig.jsonはリポジトリから削除したいと思います。

環境による差異を無くすために、`tsconfig.json`はビルド時にmcconfigが生成するものを使うよう統一したいと考えています。現在`postbuild`タスクでリンクを貼る処理をしています。

（tsconfig.jsonの中で環境変数など参照できればよかったのですが、調べた限りでは良い方法が見つかりませんでした…）

meganetaaan, 2023-01-29:
従って`generate-apidoc`をする場合は先に`build`を実行することが条件になりますね。ドキュメントも修正します。

commit a45983ce (2023-01-29): Merge pull request #117 from meganetaaan/fix/116/old-tsconfig

Fix #116 remove an old tsconfig.json
closed 2023-01-29 by meganetaaan