stack-chan/stack-chan #132: Integrate ChatGPT

meganetaaan, 2023-03-06:
The time has come! As soon as ChatGPT's API was opened, one of Stack-chan's users demonstrated a demonstration of integrating ChatGPT with Stack-chan.

https://twitter.com/robo8080/status/1631224076005179393?s=20

Mr. robo8080 created it using Arduino stacks such as M5Stack-Avatar.
"A robot that can talk to users" is the destiny of Stack-chan, and we would like to achieve this in our repository as well.

First, it is good to start with providing a thin wrapper for the ChatGPT Web API, followed by implementing a DialogManager.

The following functions are probably already present in existing dialogue systems and are desirable to implement:

- Easily set the context (system message in ChatGPT)
- Context presets to output extended syntax to change the generation method and facial expressions of Stack-chan
- Manage the user's and assistant's speech history
- Switch multiple contexts according to rules.

meganetaaan, 2023-04-18:
Introduction of ChatGPT demo on Hackaday
https://hackaday.io/project/181344-stack-chan-javascript-driven-super-kawaii-robot/log/217588-chatgpt-x-stack-chan-act2

stc1988, 2024-08-24:
We have implemented integration with several AI assistant services, including ChatGPT. If there are any remaining tasks, can we close this issue by creating a separate issue for them?

meganetaaan, 2024-10-06:
The following features have already been implemented:

- Easily setting the context (system message in ChatGPT)
- Managing the conversation history between the user and assistant

The following features could be implemented using Function Calling, as discussed in [Stack-chan Issue #293](https://github.com/stack-chan/stack-chan/issues/293):

- Context presets that output extended syntax for changing Stack-chan’s generation methods and facial expressions
- Switching between multiple contexts based on specific rules

closed 2024-10-06 by meganetaaan