stack-chan/stack-chan #135: M5Stack Core2がエラーと再起動を繰り返す

meganetaaan, 2023-03-11:
**不具合の概要**

M5Stack Core2がエラーと再起動を繰り返す。

https://twitter.com/tomolld/status/1634547949618495488

**再現手順**
不具合を再現できる手順を記述してください
1. `npm run debug --target=esp32/m5stack_core2`を実行
2. 下記エラーが発生する

**ログ**

xsbugのログ

```log
No Wi-Fi SSID
/root/Projects/moddable/modules/drivers/ft6206/modFt6206.c (131) # Break: FT6202: bad FT6206_REG_VENDID!
```

**環境 (次の項目を埋めてください):**

- OS: [Windows, MacOS, Linux(Ubuntu/Arch/etc.)] Docker
- IDE: [Arduino IDE, VSCode, UIFlow]
- ｽﾀｯｸﾁｬﾝのバージョン M5Bottom版キット
- 使用するM5Stackの種類 M5Stack Core2

meganetaaan, 2023-03-13:
手元の環境で試したが、再現せず。

- Ubuntu
- cfc19242e6b13ebfa2ce1858bb9f5e96d1bc0c2d（`dev/v1.0`の最新コミット）
- M5Stack Core2

Fuji-P, 2023-03-14:
すみません。同様の現象が発生したため書き込ませていただきます。

不具合の概要

M5Stack Core2がエラーと再起動を繰り返す（画面が黒いまま振動を続けた状態）。

再現手順
不具合を再現できる手順を記述してください
ビルド手順は下記リンクを参考にした
https://zenn.dev/yh1224/articles/34f8383gea793f1gf

package.jsonは下記を設定
{
    "config": {
        "driver": {
            "type": "pwm",
            "pwmPan": 13,
            "pwmTilt": 14
        }
    }
}

npm run debug --target=esp32/m5stack_core2を実行
xsbugのログ
→ログが出力されませんでした。Macだから？

OS：MacOS Ventura 13.2.1（M1）MacBook Air　
IDE：CLI（ModdableSDK）
ｽﾀｯｸﾁｬﾝのバージョン：M5Bottom版キット
使用するM5Stackの種類：M5Stack Core2

Fuji-P, 2023-03-15:
こちらの[リンク](https://twitter.com/lovyan03/status/1635876865113231361?s=20)のコメントみて
`if ((1 != data[0]) && (17 != data[0])) xsUnknownError("bad FT6206_REG_VENDID");`
に変更したところ、上記の環境で無事起動することができました。
取り急ぎ報告まで

meganetaaan, 2023-03-16:
@Fuji-P 
報告ありがとうございます！これはModdable側の修正が必要ですね。
取り急ぎModdableのリポジトリにissueを立てました。https://github.com/Moddable-OpenSource/moddable/issues/1064
差し支えなければ上記の変更を、Moddable本家にプルリクエスト出してみませんか？

Fuji-P, 2023-03-16:
@meganetaaan 
issueの件ご対応いただきありがとうございます。
プルリクエストの件、間に合うようなら夜くらいに対応したいと思います。

stc1988, 2023-03-16:
@meganetaaan 
コードは問題無いかと思いますが、issueのほうで関連するデータシートがあるか聞かれているようなので困りましたね😅

meganetaaan, 2023-03-16:
https://github.com/Moddable-OpenSource/moddable/issues/1064#issuecomment-1471367216
データシートの件補足しました。ドキュメント上は今も昔もFT6336Uを使っているらしい。

Fuji-P, 2023-03-16:
@meganetaaan 
色々補足いただきありがとうございます。
下記PRを出してみました。
https://github.com/Moddable-OpenSource/moddable/pull/1065

stc1988, 2023-03-18:
PRがマージされました。
私のModdableSDK環境を更新して、旧型のCore2の方でも問題なく動きました。

meganetaaan, 2023-03-22:
問題が解消されたためクローズします。

今後もM5Stack側のチップが新しくなる等で事象が再発する可能性があります。その際はReopenしてください。

closed 2023-03-22 by meganetaaan