stack-chan/stack-chan #136: Windows環境でnpm run buildが失敗する

stc1988, 2023-03-12:
**不具合の概要**
Windows環境において`npm run build` を実施すると以下エラーが発生します。

```
C:\Users\satt6\Projects\stack-chan\firmware>npm run build

> stack-chan@0.2.1 build
> cross-var mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json

### Error: '$(BLEMODULEPATH)/modBLEServer': directory not found!
npm ERR! code 1
npm ERR! path C:\Users\satt6\Projects\stack-chan\firmware
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c cross-var mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\satt6\AppData\Local\npm-cache\_logs\2023-03-12T13_27_52_386Z-debug.log
```
**環境 (次の項目を埋めてください):**

- OS: Windows
- ｽﾀｯｸﾁｬﾝのバージョン: https://github.com/meganetaaan/stack-chan/commit/cfc19242e6b13ebfa2ce1858bb9f5e96d1bc0c2d
- 使用するM5Stackの種類: M5stack

**その他**

[package.json](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/package.json#L15) でターゲットデバイスを指定するために`${npm_config_target=esp32/m5stack}` という記載がありますが、Windowsではこの記法が使えないようです。`%`で囲う必要があります。

[cross-var](https://www.npmjs.com/package/cross-var)や[cross-conf-env](https://www.npmjs.com/package/cross-conf-env) といったパッケージを導入するとLinux/macとも統一した記載ができますが、`--target`省略時の初期値を設定する方法は分かっていません。

初期値を設定する方法がなければ、`esp32/m5stack`の場合でも明示的に指定することで解決はできます。

stc1988, 2023-03-12:
ソフトウェアの修正無しに動かすためには、`package.json`のscirptを参考に`mcconfig -d -m -p esp32/m5stack -t build stackchan\manifest.json` と直接`mcconfig`コマンドを実行することで回避できます。

washishi, 2023-03-18:
cross-env-default というのがある様です
  https://www.npmjs.com/package/cross-env-default
こちらは cross-env [環境変数名]?=[未定義の際のデフォルト値]  という書き方ができるので
それと cross-var を組み合わせて下記のように記述すればとりあえず環境依存は無くせそうです
cross-env npm_config_target?=esp32/m5stack cross-var mcconfig -d -m -p $npm_config_target -t build ./stackchan/manifest.json
もっと簡単な方法があるかもしれませんが

stc1988, 2023-03-21:
twitterの方で情報をいただいた[cross-env](https://www.npmjs.com/package/@naholyr/cross-env)の方であれば、このパッケージ単独でmacは解決はできました。

washishi, 2023-03-24:
修正完了したのプルリク送りました
  https://github.com/meganetaaan/stack-chan/pull/149
色々試しましたが結局最初の cross-env-default モジュールのみで対応しました
またついでに generate-speech-google の env も cross-env への変更と
keyファイルのパスをscriptsディレクトリに変更しています

washishi, 2023-03-24:
あ・・Isuue番号間違えてる・・・一転取り消して再度送ります

washishi, 2023-03-24:
すいません送りなおしました
https://github.com/meganetaaan/stack-chan/pull/150

stc1988, 2023-03-25:
対応ありがとうございます。macても見てみますね

stc1988, 2023-03-25:
macでもbuildおよびpostbuildが引数なし（デフォルト値および）とesp32/m5stack_core2両方でエラー無く動作することを確認しました。

washishi, 2023-03-25:
package.jsonの依存は無くなったのですがWindowsでxs-dev使って確実にビルドできる方法が見つからずまだ試行錯誤中・・・

stc1988, 2023-03-25:
私の方でもwindows環境で確認可能なので、事象共有いただければ見ることできます

meganetaaan, 2023-03-27:
私のLinux（Ubuntu 20.04）環境でもビルドできました。ありがとうございます！PRを取り込みました。

xs-devのWindowsサポートは[まだベータ版なんですね](https://hipsterbrown.github.io/xs-dev/en/introduction/)。
ドキュメントに追記したほうがいいかも。

washishi, 2023-03-27:
ｽﾀｯｸﾁｬﾝファームウエアビルド時のエラーは下記[666/1144]で始まるところからです
証明書の処理のところでpythonのモジュールでエラーになってます
最初モジュールがないのか？と思いましたがモジュールはある様
手動ではESP-IDF 4.4.2でも正常にビルドできました
　(Recommend using ESP-IDF v4.4.3 (found v4.4.2) って出るけど)
Merged 1 ELF section
Successfully created esp32 image.
Generated C:/Users/stackchan/xs-dev/moddable/build/tmp/esp32/m5stack/debug/stackchan/xsProj-esp32/build/bootloader/bootloader.bin
[97/97] cmd.exe /C "cd /D C:\Users\stackchan\xs-dev\moddable\build\tmp\esp32\m5stack\debug\stackchan\xsProj-esp32\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/stackchan/xs-dev/esp32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/stackchan/xs-dev/moddable/build/tmp/esp32/m5stack/debug/stackchan/xsProj-esp32/build/bootloader/bootloader.bin"
Bootloader binary size 0x4470 bytes. 0x2b90 bytes (39%) free.
[666/1144] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle C:/Users/stackchan/xs-dev/moddable/build/tmp/esp32/m5stack/debug/stackchan/xsProj-esp32/build/esp-idf/mbedtls/x509_crt_bundle
cmd.exe /C "cd /D C:\Users\stackchan\xs-dev\moddable\build\tmp\esp32\m5stack\debug\stackchan\xsProj-esp32\build\esp-idf\mbedtls && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/stackchan/xs-dev/esp32/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Users/stackchan/xs-dev/esp32/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem C:/Users/stackchan/xs-dev/esp32/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q"
The cryptography package is not installed.Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Traceback (most recent call last):
  File "C:/Users/stackchan/xs-dev/esp32/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 36, in <module>
    from cryptography import x509
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\cryptography\x509\__init__.py", line 6, in <module>
    from cryptography.x509 import certificate_transparency
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\cryptography\x509\certificate_transparency.py", line 10, in <module>
    from cryptography.hazmat.bindings._rust import x509 as rust_x509
ImportError: DLL load failed while importing _rust: %1 は有効な Win32 アプリケーションではありません。
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
NMAKE : fatal error U1077: 'python %IDF_PATH%\tools\idf.py -n build -D mxDebug=1 -D INSTRUMENT=1 -D TMP_DIR="C:\Users\stackchan\xs-dev\moddable\build\tmp\esp32\m5stack\debug\stackchan" -D SDKCONFIG_HEADER="C:\Users\stackchan\xs-dev\moddable\build\tmp\esp32\m5stack\debug\stackchan\xsProj-esp32\build\config\sdkconfig.h" -D CMAKE_MESSAGE_LOG_LEVEL=ERROR -D DEBUGGER_SPEED=460800 -D ESP32_SUBCLASS=esp32' : リターン コード '0x2'

stc1988, 2023-03-28:
```
手動ではESP-IDF 4.4.2でも正常にビルドできました
　(Recommend using ESP-IDF v4.4.3 (found v4.4.2) って出るけど)
```

私の方ではESP-IDF v4.4.4ですが、`npm run build` は成功していました。
手動というのはnpm scriptではなく、`mcconfig`を直接実行した理解でよろしいでしょうか。

washishi, 2023-04-01:
証明書の処理のところでpythonのモジュールでエラーになる件
Pythonの違いが関係してるようです
環境構築にxs-devを使うとidf-installer2.15が動いてPython 3.8.7を取得しそれが常に使われる
手動で環境構築した場合は別途インストールしているPython3.10が使われる

stc1988, 2023-04-01:
別事象ですが、私も最近macでpython3の新しいのを入れるとビルドにこけました。

事象としては、stack-chanとも関係が無くなってきたので、一旦本issueはクローズしますね

closed 2023-04-01 by stc1988