stack-chan/stack-chan #139: ファームウェアを入れるとスタックチャンの顔が表示されるが何も動かない

Fuji-P, 2023-03-16:
**不具合の概要**
ファームウェアを入れるとスタックチャンの顔が表示されるが何も動かない

**再現手順**
Core2にファームウェアを導入

**想定する挙動**
「ランダムに動く」が正だと思われる
https://twitter.com/fuji_producer/status/1636011112587214848?s=20

**ログ**
なし

**スクリーンショット**
https://twitter.com/fuji_producer/status/1635994410079879170?s=20

**環境 (次の項目を埋めてください):**
OS：MacOS Ventura 13.2.1（M1）MacBook Air　
IDE：CLI（ModdableSDK）
ｽﾀｯｸﾁｬﾝのバージョン：M5Bottom版キット
使用するM5Stackの種類：M5Stack Core2（3/5頃購入）

**その他**
[設定例: ｽﾀｯｸﾁｬﾝ M5Bottom版キットを動かす](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/flashing-firmware_ja.md)と[ｽﾀｯｸﾁｬﾝ で始める M5Stack](https://zenn.dev/yh1224/articles/34f8383gea793f1gf)の記述で気になったのですが、
サーボの設定はstack-chan/firmware/stackchan/manifest.jsonに記載するのか、stack-chan/firmware/stackchan/manifest_local.jsonに書き込むのかがわかりませんでした。

manifest.jsonの
```
    "platforms": {
        "esp32/m5stack_core2": {
            "include": ["./manifest_8mb_flash.json"],
            "config": {
                "driver": {
                    "pwmPan": 19,
                    "pwmTilt": 27
                },

```
を
```
    "platforms": {
        "esp32/m5stack_core2": {
            "include": ["./manifest_8mb_flash.json"],
            "config": {
                "driver": {
                    "type": "pwm",
                    "pwmPan": 19,
                    "pwmTilt": 27
                },

```
とするのか

manifest_local.jsonの
```
        "driver": {
            "type": "scservo",
            "panId": 1,
            "tiltId": 2,
            "offsetTilt": 0,
            "offsetPan": 0
        }

```
を
```
        "driver": {
            "type": "pwm",
            "pwmPan": 13,
            "pwmTilt": 14
        }

```
とするのか

meganetaaan, 2023-03-17:
報告ありがとうございます。
確認したところ、ドキュメントの記載やマニフェストファイルに2点ミスがありました。

**①manifest.jsonの値が（意図せず）優先されている**

[公式ドキュメント](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/flashing-firmware_ja.md#%E8%A8%AD%E5%AE%9A%E5%A4%89%E6%9B%B4)では`manifest_local.json`の値を変えるように指示していますが、
現状では`manifest_local.json`に記載した`pwmPan`や`pwmTilt`の設定が`manifest.json`の値で上書きされてしまいます。

回避策として`manifest.json`の値を書き換えると反映されます。

**②設定例が`Port.C`のピン番号になっている**

```
                    "pwmPan": 19,
                    "pwmTilt": 27
```

このピン番号はM5Stack Core2の`Port.C`を使う場合の設定例でした。
スクリーンショットのように`Port.A`（M5Stack Core2本体側面の赤い穴）を使う場合は下記のピン番号になります。

```
                    "pwmPan": 32,
                    "pwmTilt": 33
```
（つなぎ方によってはPanとTiltが逆になるかもです）

参考：https://raspberrypi.mongonta.com/softwares-for-stackchan/

---

まとめると、`manifest.json`の該当箇所を以下のようにすると動作すると思われます。
試してみていただけますか？

```
    "platforms": {
        "esp32/m5stack_core2": {
            "include": ["./manifest_8mb_flash.json"],
            "config": {
                "driver": {
                    "type": "pwm",
                    "pwmPan": 32,
                    "pwmTilt": 33
                },
```

Fuji-P, 2023-03-17:
上記に関して動作確認しました。
```
    "platforms": {
        "esp32/m5stack_core2": {
            "include": ["./manifest_8mb_flash.json"],
            "config": {
                "driver": {
                    "type": "pwm",
                    "pwmPan": 32,
                    "pwmTilt": 33
                },
```
→こちらでは上下の首振り

```
    "platforms": {
        "esp32/m5stack_core2": {
            "include": ["./manifest_8mb_flash.json"],
            "config": {
                "driver": {
                    "type": "pwm",
                    "pwmPan": 33,
                    "pwmTilt": 32
                },

```
→こちらでは左右の首振りとなります。
どちらが想定される動作でしょうか？

closed 2023-03-21 by Fuji-P

meganetaaan, 2023-03-23:
**①manifest.jsonの値が（意図せず）優先されている**

解決策として`manifest_local.json`から`manifest.json`をincludeするようにしました。（ https://github.com/meganetaaan/stack-chan/commit/8e9da3352883e4c58e3db2a726fb8ae39ba6a17e ）。こうすることで`manifest_local.json`の設定が最も優先されるようになります。