stack-chan/stack-chan #140: Mac devContainer環境でビルドがうまくいかない

DaisukeMiyazaki, 2023-03-17:
**不具合の概要**
Mac devContainer環境でビルドがうまくいかない

**再現手順**
不具合を再現できる手順を記述してください
1. '/firmware '下でvscodeからソースコードを開く。dev containerで開きますかと聞かれるのではい -> remote desktopコンテナ内でVScodeが開く
2. root@docker-desktop:/workspaces/stack-chan/firmware# で npm install 後, m5stack core2のみ(他パーツなし)をUSB接続し、/stackchan/firmware/ 以下で `npm run deploy --target=esp32/m5stack_core2` を走らせる
3. ''./scservo/src/manifest-driver.json': manifest not found!'というエラーが発生する

**想定する挙動**
Buildが通り /build 以下ディレクトリに配置される

**ログ**
```
root@docker-desktop:/workspaces/stack-chan/firmware# npm run build  --target=esp32/m5stack_core2

> stack-chan@0.2.1 build /workspaces/stack-chan/firmware
> mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json

### Error: './scservo/src/manifest-driver.json': manifest not found!
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! stack-chan@0.2.1 build: `mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the stack-chan@0.2.1 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2023-03-17T07_08_19_669Z-debug.log

```
```
0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   '/usr/bin/node',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'run',
1 verbose cli   'build',
1 verbose cli   '--target=esp32/m5stack_core2'
1 verbose cli ]
2 info using npm@6.14.18
3 info using node@v14.21.3
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle stack-chan@0.2.1~prebuild: stack-chan@0.2.1
6 info lifecycle stack-chan@0.2.1~build: stack-chan@0.2.1
7 verbose lifecycle stack-chan@0.2.1~build: unsafe-perm in lifecycle true
8 verbose lifecycle stack-chan@0.2.1~build: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/workspaces/stack-chan/firmware/node_modules/.bin:/opt/esp/idf/components/esptool_py/esptool:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/vscode/vscode-server/bin/linux-x64/5e805b79fcb6ba4c2d23712967df89a089da575b/bin/remote-cli:/opt/esp/idf/components/esptool_py/esptool:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.23.1/bin:/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:/opt/esp/python_env/idf4.4_py3.8_env/bin:/opt/esp/idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/root/Projects/moddable/build/bin/lin/release
9 verbose lifecycle stack-chan@0.2.1~build: CWD: /workspaces/stack-chan/firmware
10 silly lifecycle stack-chan@0.2.1~build: Args: [
10 silly lifecycle   '-c',
10 silly lifecycle   'mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json'
10 silly lifecycle ]
11 silly lifecycle stack-chan@0.2.1~build: Returned: code: 1  signal: null
12 info lifecycle stack-chan@0.2.1~build: Failed to exec build script
13 verbose stack Error: stack-chan@0.2.1 build: `mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:400:28)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:400:28)
13 verbose stack     at maybeClose (internal/child_process.js:1088:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:296:5)
14 verbose pkgid stack-chan@0.2.1
15 verbose cwd /workspaces/stack-chan/firmware
16 verbose Linux 5.10.124-linuxkit
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "build" "--target=esp32/m5stack_core2"
18 verbose node v14.21.3
19 verbose npm  v6.14.18
20 error code ELIFECYCLE
21 error errno 1
22 error stack-chan@0.2.1 build: `mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t build ./stackchan/manifest.json`
22 error Exit status 1
23 error Failed at the stack-chan@0.2.1 build script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

**スクリーンショット**
もしあれば、不具合を説明するためのスクリーンショットや写真を添付してください。

**環境 (次の項目を埋めてください):**

- OS: [MacOS]
- IDE: [VSCode]
- MacbookAir M1chip
- 使用するM5Stackの種類:Core2
-  dev/1.0 latest

**その他**
1. extern/manifest_extern.jsonからエラーとみられる行をコメントアウトするとビルドは完了するが、他のエラーがでる
```
./scservo/src/manifest-driver.json",
      "./rs30x/src/manifest-driver.json"
```

```
drivers/rs30x-driver.ts:1:19 - error TS2792: Cannot find module 'rs30x'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?

1 import RS30X from 'rs30x'
                    ~~~~~~~

drivers/scservo-driver.ts:1:21 - error TS2792: Cannot find module 'scservo'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?

1 import SCServo from 'scservo'
                      ~~~~~~~~~

Found 2 errors in 2 files.

Errors  Files
     1  drivers/rs30x-driver.ts:1
     1  drivers/scservo-driver.ts:1
make: *** [/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/makefile:1733: /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/modules/workspaces/stack-chan/firmware/stackchan/ble/beacon-packet.js] Error 2
make: *** Waiting for unfinished jobs....
Loading defaults file /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/sdkconfig.mc...
/tmp/confgen_tmpp22cs8zt:62 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n
-- Configuring done
-- Generating done
-- Build files have been written to: /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! stack-chan@0.2.1 debug: `mcconfig -d -m -p ${npm_config_target=esp32/m5stack} ./stackchan/manifest.json`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the stack-chan@0.2.1 debug script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
```
2. m1(arm64)ホスト上で、amd64イメージからできたコンテナで動いていることが原因？
[image]

stc1988, 2023-03-19:
M2 macから同様dev containerで実施してみましたが、ビルド成功しました。

エラーログからは`/workspaces/stack-chan/firmware/extern/scservo/src/manifest-driver.json` が存在しないことになっていますが、このファイルは実際に存在していますでしょうか。

DaisukeMiyazaki, 2023-03-21:
@stc1988 ありがとうございます。本issue解決しました、ご指摘ただいた通り`/workspaces/stack-chan/firmware/extern/scservo/src/manifest-driver.json` が存在しておらず、 git clone 時 --recursiveオプションを入れ忘れていたことが原因でした...そのため以下トップ階層で

/stack_chan
`$ git submodule update --init --recursive`
で足りていなかった分をすべてcloneし ビルドし直したところ、上手くビルドできました。

DaisukeMiyazaki, 2023-03-21:
@stc1988 ただ deploy時にUSB接続したm5stack_core2が認識されず...試した手順含め、こちら別issueで扱おうと思います。

closed 2023-03-21 by DaisukeMiyazaki

meganetaaan, 2023-03-21:
解決できてよかったです！

`--recursive`オプション忘れちゃいますよね…。gitサブモジュールは初心者の方に易しくないので、このリポジトリにソースコードを集約して廃止したいと考えています。

参考：https://github.com/meganetaaan/stack-chan/issues/94