stack-chan/stack-chan #144: m1mac上のdevcontainerからdeployが上手くいかない

DaisukeMiyazaki, 2023-03-21:
**不具合の概要**
m5stack_core2hrdevcontainerからdeployがうまくいかない

**再現手順**
不具合を再現できる手順を記述してください
1. devcontainer 中で`npm run build --target=esp32/m5stack_core2`の後で、`npm run deploy --target=esp32/m5stack_core2`を実行する
2. 
A fatal error occurred: Failed to connect to ESP32: No serial data received.というエラーが発生する

**想定する挙動**
deployが上手く走り、hostがm5stack_core2へ書き込まれる

**ログ**

```
root@docker-desktop:/workspaces/stack-chan/firmware# npm run deploy --target=esp32/m5stack_core2

> stack-chan@0.2.1 deploy /workspaces/stack-chan/firmware
> mcconfig -d -m -p ${npm_config_target=esp32/m5stack} -t deploy ./stackchan/manifest.json

/opt/esp/idf/tools/idf.py
/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/makefile:2127: warning: overriding recipe for target '/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/ftimage.h.xsi'
/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/makefile:2058: warning: ignoring old recipe for target '/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/ftimage.h.xsi'
# uploading to esp32
[1/5] cd /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/esp-idf/esptool_py && /opt/esp/python_env/idf4.4_py3.8_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/partition_table/partition-table.bin /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/xs_esp32.bin
xs_esp32.bin binary size 0x1df790 bytes. Smallest app partition is 0x200000 bytes. 0x20870 bytes (6%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf4.4_py3.8_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/bootloader/bootloader.bin
Bootloader binary size 0x4420 bytes. 0x2be0 bytes (39%) free.
[2/3] cd /opt/esp/idf/components/esptool_py && /opt/esp/tools/cmake/3.23.1/bin/cmake -D IDF_PATH="/opt/esp/idf" -D SERIAL_TOOL="/opt/esp/python_env/idf4.4_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build" -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/ttyAMA0 -b 1500000 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x1000 bootloader/bootloader.bin 0x10000 xs_esp32.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3.2
Serial port /dev/ttyAMA0
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:56 (message):
  /opt/esp/python_env/idf4.4_py3.8_env/bin/python
  /opt/esp/idf/components/esptool_py/esptool/esptool.py --chip esp32 failed

FAILED: CMakeFiles/flash /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/CMakeFiles/flash 
cd /opt/esp/idf/components/esptool_py && /opt/esp/tools/cmake/3.23.1/bin/cmake -D IDF_PATH="/opt/esp/idf" -D SERIAL_TOOL="/opt/esp/python_env/idf4.4_py3.8_env/bin/python /opt/esp/idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build" -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Executing action: flash
Running ninja in directory /root/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build
Executing "ninja flash"...
```

**スクリーンショット**
host OS側では m5stack_core2 は /dev/以下に認識されている (UARTでは出てこないがm5burner, arduinoではこのデバイスポートから接続できる)
[image]

container側では /dev/以下にttyが反映されていない
[image]

元々自動で登録される/dev/ttyAMA0 には日時が更新されている模様だが、こちらがUSBserial port用なのか不明
[image]

**環境 (次の項目を埋めてください):**

- OS: [MacOS , m1 chip]
- IDE: [VSCode, UIFlow]
- M5Stack_core2

**その他**

- [実際のリンクではなくsymlinkしてdocker runすれば上手くいく](https://itecnote.com/tecnote/linux-exposing-a-tty-device-in-a-docker-container-with-docker-for-mac/
)？dev contaninerでないので、devcontainer.json の mounts: source=/dev/, target=/dev/ に追加が必要？ 

- [mac0Sから tty serial deviceをexposeしても上手くいかない? ](https://github.com/docker/for-mac/issues/900)
- [rasberrypiならうまくいく？](https://geek.tacoskingdom.com/blog/63)

#140 の続きです。

stc1988, 2023-03-28:
確かに現状のmac+Dockerだとコンテナ側からホスト(mac)側のデバイスを見るのが難しいようです。

ESP-IDFの[Remote Serial Ports](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html#usage)を試してみると`npm run debug`で書き込み自体はできているのですが、その後のマイコン側へのリセットが失敗する（手動で電源off/onすればOK）ので、開発体験としてはよくなさそうです。

あとmac+Dockerだとmac単独環境に比べてビルドが非常に遅いので、個人的にはmac上で環境構築いただくのがおすすめです。

@meganetaaan 
Windows+DockerでもCOMポートが見えないかと思いますので、DockerコンテナはLinuxホスト推奨とした方が良いかもしれません。

meganetaaan, 2023-04-01:
@stc1988 ありがとうございます！DockerはLinux以外では非推奨の旨追記しておきます。
@DaisukeMiyazaki お手数ですが[xs-devを使う手順](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/getting-started_ja.md#xs-dev%E3%82%92%E4%BD%BF%E3%81%86%E6%8E%A8%E5%A5%A8)を試していただけますか？うまく行かない場合はまたぜひご質問ください🙆

DaisukeMiyazaki, 2023-04-02:
@meganetaaan @stc1988 ご対応ありがとうございます！macローカル環境でxs-devを使う手順で実行したところ、upload portの再設定などCLI上で必要でしたが、うまくいきました！ありがとうございました！

closed 2023-04-02 by DaisukeMiyazaki