stack-chan/stack-chan #154: 本リポジトリに掲載いただいているModの期待値を教えていただきたい（可能であればRead meに記載します）

Fuji-P, 2023-04-02:
**既存の機能に問題がある場合、その説明**
現在本リポジトリに掲載いただいているModの期待値を教えていただきたい（可能であればRead meに記載します）

**あなたが望む解決策**。
現在下記環境でLook atのModを導入したのだが、どのように動作させれば良いかがわからない。
→xsbug起動後適当にボタンを押すとlooking atとapply poseを呼び出すようになるが、looking atの呼び出し頻度が低そうで首振り頻度が低い様子
また、Mod導入後はPCに接続せずスタンドアロンで機能を使用できるのかが知りたい
→xsbugでデバッグしつつでないと動作しない様子？

環境:
OS：MacOS Ventura 13.2.1（M1）MacBook Air　
IDE：CLI（ModdableSDK）
ｽﾀｯｸﾁｬﾝのバージョン：M5Bottom版キット
使用するM5Stackの種類：M5Stack Core2（3/5頃購入）

stc1988, 2023-04-12:
#133 同様に動画や画像があった方が望ましいですね。
また、多くのmodはTTSなど外部システムと連携するものが多いので、動作に何が必要かの記載もあると良さそうです。

meganetaaan, 2026-09-17:
起票時の個別サンプル構成は現在の構成へ移行しており、Look at に相当する例は look_around など現行 MOD として整理されています。各 MOD の README 整備という一般課題は #255 で追跡しているため、本 Issue は obsolete / 集約済みとしてクローズします。

closed 2026-09-17 by meganetaaan