stack-chan/stack-chan #168: アプリ領域のパーティションサイズが不足している

stc1988, 2023-05-14:
**不具合の概要**

新規機能追加のために` "$(MODDABLE)/examples/pins/audioout/elevenlabs-stream/manifest_elevenlabsstreamer.json"`などをincludeしてビルドするとエラーとなる

**ログ**

```
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/esp-idf/esptool_py/CMakeFiles/app_check_size
cd /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/esp-idf/esptool_py && /Users/satoshi/.espressif/python_env/idf4.4_py3.8_env/bin/python /Users/satoshi/.local/share/esp32/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/partition_table/partition-table.bin /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/xs_esp32.bin
Error: app partition is too small for binary xs_esp32.bin size 0x208f80:
  - Part 'factory' 0/0 @ 0x10000 size 0x200000 (overflow 0x8f80)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

**その他**

[customPartitions.csv](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/stackchan/sdkconfig/customPartitions.csv) を使用する場合(core2)には、アプリ領域が2MBとなっているためサイズ不足していると思われる。

meganetaaan, 2023-05-16:
報告ありがとうございます。MODの領域をなるべく確保したくてアプリ領域を必要最小限にしていました。
ホスト側に組み込むモジュールを増やすに従い、都度ホスト領域を拡大する方針としたいと思います。

stc1988, 2023-05-18:
必要に応じて都度拡大で承知しました。

若干別件ですが、`sdkconfig.defaults`でFLASHのサイズを拡張していますが、[core2のもの](https://github.com/Moddable-OpenSource/moddable/blob/fece720ad1456ad571fa4582560c3ecf7d9d7400/build/devices/esp32/targets/m5stack_core2/sdkconfig/sdkconfig.defaults)はFLASH拡張とPSRAMを使用するように変更されたので、stack-chan側での変更は不要そうです。

stc1988, 2023-09-21:
#171 がマージされ容量を拡張したのでクローズします

closed 2023-09-21 by stc1988