stack-chan/stack-chan #177: M5Stack CoreS3をサポートする / Support M5Stack CoreS3

meganetaaan, 2023-06-03:
[M5Stack CoreS3](https://docs.m5stack.com/en/core/CoreS3)はデュアルマイクやカメラなど、コミュニケーションロボットのインタフェースに使えるペリフェラルを多数備えています。ぜひｽﾀｯｸﾁｬﾝでサポートしたいです。

[M5Stack CoreS3](https://docs.m5stack.com/en/core/CoreS3) is equipped with numerous peripherals that can be used for the interface of communication robots, such as dual microphones and a camera. I would definitely like to support it with Stack-chan.

meganetaaan, 2023-06-11:
https://github.com/meganetaaan/moddable/tree/feature/m5stack_cores3 で実装着手しました。

LCDが初期化できることを確認。`/examples/piu/balls`が動作しています。

https://github.com/meganetaaan/stack-chan/assets/1918755/b74da18e-934b-4a08-9206-95595f30032b

meganetaaan, 2023-06-25:
Moddable本家にPRを出しました。

https://github.com/Moddable-OpenSource/moddable/pull/1165

meganetaaan, 2023-07-21:
7月のリリースに取り込まれました。

https://github.com/Moddable-OpenSource/moddable/releases/tag/3.9.7

デュアルマイクやカメラなど未実装のペリフェラルが多数あるため、issueはまだopenのままにします。

meganetaaan, 2024-11-16:
もう1年以上経つのか :roll_eyes: wow

今の状況です↓

カメラ … Moddable [v5.2.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/5.2.0) でカメラ機能がサポートされました。
デュアルマイク … https://github.com/meganetaaan/moddable/tree/feature/cores3_mic で作業中。 https://github.com/m5stack/M5Unified から初期化処理を移植したが反応せず。設定を確認中。

stc1988, 2024-12-22:
マイクについては別途Moddable側でissueが作成されたので、そちらの方で進んでいくかも

https://github.com/Moddable-OpenSource/moddable/issues/1439