stack-chan/stack-chan #183: ファームウェアビルド時のエラー

taisan11, 2023-07-16:
**不具合の概要**
ファームウェアをビルドした際にエラーが発生する

**再現手順**
不具合を再現できる手順を記述してください
1.`npm i`をしてvisualstudioを入れ`npm run setup`はvisualstudiocommandpalletv17.6.5で、`npm run setup -- --device=esp32`は`npm run setup`で出てきた新しいコマンドプロンプトで実行し、
1. [これ](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/flashing-firmware_ja.md)の`npm run build --target=esp32/m5stack_core2`をESP-IDF 5.1 CMDで実行する
2. 下のようなエラーが発生する

**ログ**

```log
> stack-chan@0.2.1 build
> cross-env npm_config_target?=esp32/m5stack cross-env-shell mcconfig -d -m -p \$npm_config_target -t build ./stackchan/manifest_local.json

C:\Espressif\tools\idf-exe\1.0.3\idf.py.exe
C:\Espressif\frameworks\esp-idf-v5.1\tools\idf.py
*** Update required to ESP-IDF v4.4.3 (found v5.1)
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md
"Expected ESP IDF v4.4.3, found v5.1"
NMAKE : fatal error U1077: 'python C:\Users\ユーザー名\xs-dev\moddable\build\devices\esp32\xsProj-esp32\versionCheck.py v4.4.3 v5.1 || (echo "Expected ESP IDF v4.4.3, found v5.1" && exit 1)' : リターン コード '0x1'
Stop.
```

**環境:**

- OS: Windows11
- IDE: ESP-IDF?
- ｽﾀｯｸﾁｬﾝのバージョン:dev/v1.0
- M5Stack Core2

stc1988, 2023-07-20:
ESP IDFの`v5.1`がインストールされているようですが、現在Moddableで対応しているのは`v4.4.3`です。
インストール方法については[Moddable公式ドキュメント](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#windows)を参照ください

taisan11, 2023-07-22:
ESP-IDFv4.4.3のコマンドプロンプト内で`C:\Users\kappy\xs-dev\Moddable.bat`を実行し`npm run build --target=esp32/m5stack_core2`を実行したところ以下のようなエラーが発生します
```
-- Configuring incomplete, errors occurred!
See also "C:/Users/user/xs-dev/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
NMAKE : fatal error U1077: 'python %IDF_PATH%\tools\idf.py -n reconfigure -D SDKCONFIG_DEFAULTS=C:/Users/user/xs-dev/moddable/build/tmp/esp32/m5stack_core2/debug/stackchan/xsProj-esp32/sdkconfig.mc -D SDKCONFIG_HEADER="C:\Users\user\xs-dev\moddable\build\tmp\esp32\m5stack_core2\debug\stackchan\xsProj-esp32\build\config\sdkconfig.h" -D CMAKE_MESSAGE_LOG_LEVEL=ERROR -D DEBUGGER_SPEED=1500000 -D IDF_TARGET=esp32 -D ESP32_SUBCLASS=esp32 -D SDKCONFIG_DEFAULTS=C:\Users\user\xs-dev\moddable\build\tmp\esp32\m5stack_core2\debug\stackchan\xsProj-esp32\sdkconfig.mc' : リターン コード '0x2'
Stop.
```
生成されたログは以下の通りです
https://gist.github.com/taisan11/b621128f20267c9a0b9f16ff1df9625d

stc1988, 2023-08-01:
確認遅くなりすみません。
いただいたログからも手がかりがないので`npm run doctor`を実行して環境が構築できているか確認いただけますでしょうか

taisan11, 2023-08-02:
`npm run doctor`を実行してみましたが特におかしい点はありませんでした
以下実行ログです
```
stack-chan environment info:
4bf78b32f11bdd3457a3826fd247a2dfeb99e808
C:/Users/kappy/Documents/stack-chan
xs-dev environment info:
  CLI Version                0.20.0
  OS                         Windows_NT
  Arch                       x64
  NodeJS Version             v18.16.0 (C:\Program Files\nodejs\node.EXE)
  Python Version             3.8.7 (C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.EXE)
  Moddable SDK Version       3.9.2 (C:\Users\kappy\xs-dev\moddable)
  Supported target devices   win, esp32
  ESP32 IDF Directory        C:\Users\kappy\xs-dev\esp32\esp-idf

If this is related to an error when using the CLI, please create an issue at "https://github.com/hipsterbrown/xs-dev/issues/new" with the above info.
```

stc1988, 2023-08-02:
環境構築はxs-devを使って実施してそうですね。私のほうでも試してみますが、Windows環境のxs-devはサポート状況が完全でなく、動作しない可能性もあります。

[Windows環境においては[公式の手順](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md#win-instructions)を実施し他が早回りかもしれません。

@washishi  本件のような事象に遭遇したことはありますでしょうか

washishi, 2023-08-02:
Windowsの場合 xs-dev というか ESP-IDF Windows InstallerでPython3.8.7が入りそれが使われるのですがそのPythonに問題がある様でｽﾀｯｸﾁｬﾝのファームウエアのビルドは失敗します
なので利用するPythonを変更する必要があります

下記メモを作ってますので参照してみてください
[Windowsでのxs-devを利用したmodableビルド環境構築](https://docs.google.com/document/d/1d9tJtlop8iQA_WK1ZvKx-tOoo5K1ntlmiUGx0NkkASc/edit?usp=sharing)

利用するPythonの変更については「・ESP-IDF Tools で利用するPythonの変更」のところで書いてます

washishi, 2023-08-02:
【参考】ビルドで失敗する件のメッセージの詳細はこちらで書いてましたね
https://github.com/meganetaaan/stack-chan/issues/136#issuecomment-1485041097

closed 2025-04-25 by taisan11