stack-chan/stack-chan #215: Stack-chan fails to build on Moddable v4.2.0

meganetaaan, 2023-10-03:
**Describe the bug**
Stack-chan fails to build on Moddable v4.2.0

**To Reproduce**
Steps to reproduce the behavior:
1. Update Moddable SDK to v4.2.0
2. Update ESP-IDF to v5.1.1
3. run `npm run debug`

**Logs**
If any compile error occurs, add a full compile log.

`pins/servo` module seems to be incompatible with the latest ESP-IDF (v5.1.1)

````log
# cc modServo.c.o
# cc crc.c.o
/home/sskw/.local/share/moddable/modules/pins/servo/esp32/modServo.c: In function 'xs_servo':
/home/sskw/.local/share/moddable/modules/pins/servo/esp32/modServo.c:58:18: error: 'ledc_timer_config_t' has no member named 'bit_num'
   58 |                 .bit_num = LEDC_TIMER_14_BIT,
      |                  ^~~~~~~
make: *** [/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/makefile:2591: /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/modServo.c.o] エラー 1
make: *** 未完了のジョブを待っています....
# cc modules.c.o
````

<details>
<summary>complete log</summary>

```
[sskw]$ npm run debug target=esp32/m5stack_core2            

> stack-chan@0.2.1 debug /work/ghq/github.com/meganetaaan/stack-chan/firmware
> cross-env npm_config_target?=esp32/m5stack cross-env-shell mcconfig -d -m -p \$npm_config_target ./stackchan/manifest_local.json "target=esp32/m5stack_core2"

mcconfig: mod partition of size 0x40000 created from factory app partition
mcconfig: file storage partition of size 0x10000 created from factory app partition
/home/sskw/.local/share/esp32/esp-idf-v5.1.1/tools/idf.py
/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/makefile:2376: 警告: ターゲット '/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/ftimage.h.xsi' のためのレシピを置き換えます
/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/makefile:2307: 警告: ターゲット '/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/ftimage.h.xsi' のための古いレシピは無視されます
# bles2gatt bleservices
Using recommended ESP-IDF v5.1.1
# xsc Resource.xsb
# xsc ili9341.xsb
# xsc m5button.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc modules.xsb
# xsc crc.xsb
# xsc net.xsb
# xsc socket.xsb
# xsc sntp.xsb
# xsc wifi.xsb
# xsc preference.xsb
# xsc structuredClone.xsb
# xsc deepEqual.xsb
# xsc bleserver.xsb
# xsc btutils.xsb
# xsc gap.xsb
# xsc gapwhitelist.xsb
# xsc sm.xsb
# xsc bleclient.xsb
# xsc connection.xsb
# xsc gatt.xsb
# xsc uartserver.xsb
# xsc stk-server.xsb
# xsc crypt.xsb
# xsc base64.xsb
# xsc hex.xsb
# xsc ec.xsb
[... 533 lines cut ...]
# cc structuredClone.c.o
# cc spi.c.o
# cc _i2c.c.o
# cc digitalbank.c.o
# cc builtinCommon.c.o
# cc xs6wifi.c.o
# cc modsntp.c.o
# cc modSocket.c.o
# cc modResolve.c.o
# cc modNet.c.o
# cc modServo.c.o
# cc crc.c.o
/home/sskw/.local/share/moddable/modules/pins/servo/esp32/modServo.c: In function 'xs_servo':
/home/sskw/.local/share/moddable/modules/pins/servo/esp32/modServo.c:58:18: error: 'ledc_timer_config_t' has no member named 'bit_num'
   58 |                 .bit_num = LEDC_TIMER_14_BIT,
      |                  ^~~~~~~
make: *** [/home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/makefile:2591: /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/debug/stackchan/modServo.c.o] エラー 1
make: *** 未完了のジョブを待っています....
# cc modules.c.o
```
</details>

- OS: Linux
- Stack Chan Version: latest `dev/v1.0` branch
- Type of M5Stack to use: M5Stack Core2

stc1988, 2023-10-03:
https://github.com/Moddable-OpenSource/moddable/issues/1088

I can't confirm it right now, but hasn't this problem been solved yet?

meganetaaan, 2023-10-03:
I've found that another commit rolled back your fix. I made a PR to get them back.
https://github.com/Moddable-OpenSource/moddable/pull/1225

meganetaaan, 2023-10-03:
I've found several issues that block firmware build.
As a workaround, I'm going to fix the version of ModdableSDK in `npm run setup` script as below.

```
    "setup": "xs-dev setup --target-branch 4.1",
```

stc1988, 2024-08-24:
We have resolved various issues and are currently using the latest version of Moddable. Although there is an issue with the serial servo not working in the latest version, can we create a separate issue for that and close this issue?