stack-chan/stack-chan #237: m5stack core2 v1.1でファームウェアが全く機能しません

RChikamura, 2024-02-01:
**不具合の概要**
ファームウェアをビルドし、書き込んでみたのですが、画面に何もつかず、モーターも反応しません
ターミナルやデバッガーを見ても、目立ったエラーコードは特に出ていません。

**再現手順**
node js および npm の最新版を導入した環境で、
https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/getting-started_ja.md
https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docs/flashing-firmware_ja.md
のコードを用いて環境構築をし、ファームウェアを書き込みました。

**環境:**

- OS: Windows および Ubuntu (どちらで試してもダメでした)
- ESP-IDF v5.1.2 および v4.4.3 (v4.4.3はUbuntuでのみテスト)
- ｽﾀｯｸﾁｬﾝのバージョン v1.0
- 使用するM5Stackの種類 m5stack core2 v1.1

meganetaaan, 2024-02-01:
報告ありがとうございます！Core2 v1.1はModdable側がまだ対応しておらず、改修が必要です。
私も手元にテスト用機体はありますが手が回っていない状況でした🙇
今週中を目処に、必要な改修箇所について確認しますね。

RChikamura, 2024-02-18:
私のほうでも独自に調査し、改修をしていたのですが、前進があったので共有いたします
Core2のsetup-target.js, manifest.jsonと、モジュールaxp2101.jsを改修することにより、初期化処理が一部進行するようになった模様です。ただし、タッチパネルのセットアップに失敗しているようで、スタックちゃんの起動までは至らず、下部LEDのみが点灯している状態となります。
https://github.com/RChikamura/Moddable_Repair_RChikamura

RChikamura, 2024-02-25:
さらに修正を加え、プログラムが１回起動するところまでは持っていけましたが、２回目以降は次回正常起動ができず、わずか数秒で落ちてしまう状態になってしまいます。

問題点として、AXP2101.jsにピン設定が大幅に不足していたり、AXP192.jsから変更されていないままの部分があったりしている点が見受けられ、それが影響を及ぼしているように思えます。
https://github.com/RChikamura/Moddable_Repair_RChikamura

RChikamura, 2024-02-29:
さらに修正を加え、おそらく正常起動可能になりました。
https://github.com/RChikamura/moddable_core2v1.1_try

meganetaaan, 2024-03-07:
ありがとうございます！いまコード読んでいます。
AXP2101のDCDCやLDOなど足りていない機能を色々実装してくださったんですね。

DCDCの電圧のステップ数などの設定が[ピンによって違う](https://github.com/RChikamura/moddable_core2v1.1_try/blob/d22782d4398db27bdda5e999d454615cc70303e3/xs-dev/moddable/modules/drivers/axp2101/axp2101.js#L33)んですね。たしかに[データシート](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/K128%20CoreS3/AXP2101_Datasheet_V1.0_en.pdf)の5.5 （Electrical Characteristics）に書いてますね。

後ほど実機で動かしてみます！

ちなみにModdable本家に今回の対応をコントリビュートするのは興味ありますか？コードを他のデバイスでも動くように修正したり、プルリクエストを作ったりする感じです。慣れない点がありましたら私もサポートできればと思います。

RChikamura, 2024-03-13:
返信遅れて申し訳ございません。
このようにgithubで連携を取って開発を進めたことがなく、まだどのような作業を行えばよいかわかっていないのですが、できることがあればお手伝いしたいと思います。

meganetaaan, 2024-03-17:
私の方でも動作確認ができました！ありがとうございます！

いただいた変更を私の方でModdableに取り込んで微修正し、旧バージョンのCore2と共存して動くようにできました。

https://github.com/meganetaaan/moddable/tree/feature/m5stack_v2.1

Moddable最新版からの差分は下記のプルリクエストのとおりです。
https://github.com/meganetaaan/moddable/pull/1

このあともう少しセルフレビューしたら、RChikamuraさんを「共同編集者」としてModdable本家のほうにプルリクを出そうと思います。その際RChikamuraさんにContributor License Agreement（CLA：Moddableが指定するライセンスでコードを公開してよいですという同意書）を書いていただく必要がありますが、この段階になったらまた連絡します。CLAの詳細は↓にあります。

https://github.com/Moddable-OpenSource/moddable/blob/public/contributing.md#submitting-pull-requests

meganetaaan, 2024-03-22:
プルリク作りました！

https://github.com/Moddable-OpenSource/moddable/pull/1324

@RChikamura 前述の通り、ModdableにCLAを提出していただけますでしょうか。

https://github.com/Moddable-OpenSource/moddable/blob/public/licenses/Moddable%20CLA%20-%20individual.pdf

* PDFを印刷
* 必要事項を記入
* スキャン（私はPDFスキャン機能のついたスマホアプリを使いました）
* info@moddable.com にPDFを送付する

で完了です。ご不明点ありましたらお尋ねください :bow:

meganetaaan, 2024-03-29:
上記のプルリクがマージされました。
https://github.com/Moddable-OpenSource/moddable/pull/1324

既にModdableの`public`ブランチで利用できるほか、来月のModdable月次リリースにもこの修正が含まれる見込みです。

meganetaaan, 2024-05-02:
https://github.com/Moddable-OpenSource/moddable/releases/tag/4.6.0
4月のリリースで修正が取り込まれたので試しているんですが、LCDが付かないですね…なにか修正を間違えたかな。
デバイス本体下の青色LEDがずっと点滅しているのも気になります。

meganetaaan, 2024-05-02:
クローズします。v1.1でまだ問題があれば再オープンしてください。

closed 2024-05-02 by meganetaaan