stack-chan/stack-chan #239: Dockerイメージでのビルドに失敗する

stc1988, 2024-02-18:
[Discord](https://discord.com/channels/1095725099925110847/1095730172789006468/1208648066572484669)より

[Dockerfile](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/docker/Dockerfile)で指定されているModdabeSDKのバージョンが古く、また、[Dockerイメージ](https://hub.docker.com/layers/meganetaaan/moddable-esp32/3.9.7/images/sha256-fee16d87ccd2f98429b0cf3a4383d455648ed5decd4ad89a261a6c2eae952530?context=explore)で指定されているESP-IDFもバージョンも古いためバージョンアップが必要。

Node,jsのバージョンも14とEOLのため、20あたりまで上げた方が良さそうです。GitHub actionsも合わせてあげた方が良いです。

GitHub actions同様にxs-devで環境構築できるようにした方がイメージのメンテナンスコストは減るかもしれません。

meganetaaan, 2024-10-05:
>xs-devで環境構築

この対応方針に賛成です。xs-devの機能が充実してきたので全面的に採用して、手動での構築を辞める方向にします。

meganetaaan, 2025-01-06:
https://github.com/stack-chan/stack-chan/tree/feature/refine-container で作業中です。

`CI=true`を付けることでDockerfileのCLI環境でxs-devを用いた環境構築ができました。

`build-container.sh` でビルド、 `launch-container.sh` でコンテナを起動して、書き込みができるところまで確認しました。
（環境はUbuntu24.04のDocker）

meganetaaan, 2025-01-06:
残件

- [ ] devcontainerの起動設定を追加
- [ ] 環境構築のドキュメントを修正

WindowsとmacOSでの動作確認は有志の方にお願いしたいです :thinking:

stc1988, 2025-01-06:
MacOSでも動作確認はできますが、デバイスへの接続設定がネックだったかと思います。
Windowsはwslからのdockerなら行けるのかな🤔

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-docker-image.html

いったんはLinuxのみ確認済みでもよいかと思います

stc1988, 2025-02-02:
#311 が対応されたのでクローズします

closed 2025-02-02 by stc1988