stack-chan/stack-chan #243: 音量を設定できるようにする

stc1988, 2024-03-17:
https://github.com/meganetaaan/stack-chan/issues/204 の残課題

**Is your feature request related to a problem? Please describe.**

TTSなどでｽﾀｯｸﾁｬﾝから音声出力時の音量を変更できるようにする。

**Describe the solution you'd like**

AudioOutで音量を変更するには、[audio.enqueue](https://github.com/Moddable-OpenSource/moddable/blob/a6a71b225bd35201ec76b2c354b7595a1be2ed12/documentation/pins/audioout.md#changing-volume)時に指定する方法があるが、`xxxStreamer`内で指定しているため拡張が必要？

M5stackCoreS3の場合は[AW88298](https://github.com/Moddable-OpenSource/moddable/blob/a6a71b225bd35201ec76b2c354b7595a1be2ed12/build/devices/esp32/targets/m5stack_cores3/setup-target.js#L131-L134)から設定が可能？

**Additional context**

`manifest_json`の`config`から指定する場合や、NVSで設定値を保持する場合、 [Preferenceのキー](https://github.com/meganetaaan/stack-chan/blob/dev/v1.0/firmware/stackchan/utilities/consts.ts#L9)が未定義であるため定義が必要。

stc1988, 2024-10-22:
#291 でAudioOutインスタンスを都度生成するようになったので、そのタイミングで音量をセットすることができそう。
と同時にToneクラスも誕生したため、stack-chan一律での音量設定よりはクラスごとのインスタンス変数として音量を保持、設定できるようにした方がよい。

meganetaaan, 2025-01-06:
音量に関する要求事項として以下が考えられます

- ｽﾀｯｸﾁｬﾝを置く環境によって、後から音量を変えられるようにしたい
- 一部の音声（通知音など）は上記設定に関わらず固定の音量を与えられるようにしたい

実装方針を考えてみました

- TTSの各クラスとToneに音量セットの機能を追加する
- コンストラクタ引数にオプションvolumeを足す
- Robotクラスに音量の変数を追加する
- configに音量設定を追加する→`config.tts.volume`
- Robot#sayの第2引数で音量をオーバーライドできる

stc1988, 2025-06-07:
対対応したのでクローズします

closed 2025-06-07 by stc1988