stack-chan/stack-chan #246: WebブラウザからM5Stack CoreS3へのファームウェア書き込みに失敗する

meganetaaan, 2024-03-21:
**不具合の概要**

[Webサイトからのｽﾀｯｸﾁｬﾝファームウェア書き込み](https://meganetaaan.github.io/stack-chan/web/flash/) 機能がM5Stack CoreS3では失敗します。

**再現手順**
1. PCとM5Stack CoreS3をUSB接続する
2. https://meganetaaan.github.io/stack-chan/web/flash/ を開く
3. ターゲットデバイスに「M5Stack CoreS3」を選択する
4. "Flash Stack-chan Firmware" ボタンを押下してシリアルポートを選択する
5. "Install Stack-chan"を選択する
6. "Your ESP32S3 board is not supported"というエラーが出て書き込みに失敗する

**想定する挙動**
正しくファームウェアの書き込みができる

**ログ**

書き込み失敗するときのWeb開発者ツールのコンソールのログは下記の通り

```log
ESP Web Tools 9.4.3 by Nabu Casa; https://esphome.github.io/esp-web-tools/
install-dialog-28H-HNrD.js?module:424 Initializing Improv Serial
install-dialog-28H-HNrD.js?module:724 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-28H-HNrD.js?module:424:1355
_initialize @ install-dialog-28H-HNrD.js?module:724
install-dialog-28H-HNrD.js?module:424 esptool.js
install-dialog-28H-HNrD.js?module:424 Serial port WebSerial VendorID 0x303a ProductID 0x1001
install-dialog-28H-HNrD.js?module:424 Connecting...
install-dialog-28H-HNrD.js?module:424 .
install-dialog-28H-HNrD.js?module:424 

install-dialog-28H-HNrD.js?module:424 Detecting chip type... 
install-dialog-28H-HNrD.js?module:424 ESP32-S3
install-dialog-28H-HNrD.js?module:424 Chip is ESP32-S3
install-dialog-28H-HNrD.js?module:424 Features: Wi-Fi,BLE
install-dialog-28H-HNrD.js?module:424 Crystal is 40MHz
install-dialog-28H-HNrD.js?module:424 MAC: 48:27:e2:66:e0:34
install-dialog-28H-HNrD.js?module:424 Uploading stub...
install-dialog-28H-HNrD.js?module:424 Running stub...
install-dialog-28H-HNrD.js?module:424 Stub running...
install-dialog-28H-HNrD.js?module:424 Manufacturer: 20
install-dialog-28H-HNrD.js?module:424 Device: 4018
install-dialog-28H-HNrD.js?module:424 Detected flash size: 16MB
install-dialog-28H-HNrD.js?module:724 Uncaught (in promise) DOMException: Failed to execute 'open' on 'SerialPort': The port is already open.
    at https://unpkg.com/esp-web-tools@9.4.3/dist/web/install-dialog-28H-HNrD.js?module:724:6424
```

**環境 (次の項目を埋めてください):**

- OS: Linux
- ブラウザ: Google Chrome バージョン: 112.0.5615.165（Official Build） （64 ビット）
- ｽﾀｯｸﾁｬﾝのバージョン `dev/v1.0`最新
- 使用するM5Stackの種類 M5Stack CoreS3

meganetaaan, 2024-03-21:
ESP Web Toolsの公式サイトのライブデモ（https://esphome.github.io/esp-web-tools/）からはCoreS3に正しく書き込みができる。

ｽﾀｯｸﾁｬﾝのESP Web Toolsはバージョン9.4.3、公式サイトは10.0.1。単にバージョンを最新にすると解決するかも。

stc1988, 2024-03-21:
本issueと同様の事象が再現できました。

[esptool-js](https://github.com/espressif/esptool-js)側のアップデートかもしれないですが、ESP Web Toolsを最新にしてみるで良い気はします。

meganetaaan, 2024-03-22:
プルリク作りました。結局esp-web-tool用のマニフェストファイルに不備があり、チップファミリーの指定とブートローダーを書き込むオフセットが間違っていました。オフセットについてはビルド時の情報から自動で抽出する案もありましたがまだベタ書きですね…

commit af6bea01 (2024-03-23): Merge pull request #247 from meganetaaan/fix/web-flash-esp32s3

Fix #246 web manifest for flashing to M5Stack CoreS3
closed 2024-03-23 by meganetaaan