stack-chan/stack-chan #264: @moddable/typings パッケージを使用する

stc1988, 2024-08-16:
**既存の機能に問題がある場合、その説明**

#116にて`tsconfig.json`からModdableの標準モジュールのtypingsを相対パスで指定できない問題が発生するため、#117 で`tsconfig.json`をビルド時に生成されるものをコピーするようにしました。

しかし、[ @moddable/typings](https://www.npmjs.com/package/@moddable/typings)がnpmパッケージとしてModdableのリリースと同期して配信されるようになったため、ビルド時にコピーすることは不要になる想定です。

**あなたが望む解決策**。

`package.json`に `@moddable/typings` パッケージを追加し、`tsconfig.json`をコード管理するようにする。
また、`tsconfig.json`に`mods`や`tests`も含めることでそれらに含まれるファイルからも型定義の補完をできるようにしたいです。

stc1988, 2024-08-17:
現状の[進捗](https://github.com/stc1988/stack-chan/commit/06af7fb3aacc846fe6ed1a3551d8d115712571fe)です

方針としては
* 直接ビルドに使用するものないファイルであり、今後ファイル追加による修正を最小限にすべく、`*`を使って広く対象をマッチさせるようにしています
* `mods`以下などのJSファイルでも補完が効くように`"allowJs": true`としています。

状況としては、
* ビルドは通る([CIのアクション](https://github.com/stc1988/stack-chan/actions/runs/10433230566)は問題なし)
* `npm run lint`も通る
* `npm run generate-apidoc` が失敗する

```
npm run generate-apidoc

> stack-chan@0.2.1 generate-apidoc
> typedoc --entryPointStrategy expand --plugin typedoc-plugin-markdown --out docs/api ./stackchan

Loaded plugin typedoc-plugin-markdown
stackchan/speeches/calculate-power.js:1:33 - error TS1144: '{' or ';' expected.

1 function calculatePower(sample) @ "xs_calculatePower";
                                  ~

stackchan/speeches/calculate-power.js:1:54 - error TS1146: Declaration expected.

1 function calculatePower(sample) @ "xs_calculatePower";

typings/bleserver.d.ts:2:19 - error TS2305: Module '"btutils"' has no exported member 'Advertisement'.

2   import { Bytes, Advertisement } from 'btutils'
                    ~~~~~~~~~~~~~

typings/bleserver.d.ts:3:23 - error TS2792: Cannot find module 'gap'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?

3   import { GAP } from 'gap'
                        ~~~~~

typings/bleserver.d.ts:149:23 - error TS2792: Cannot find module 'gap'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?

149   import { GAP } from 'gap'
                          ~~~~~

typings/bleserver.d.ts:203:23 - error TS2792: Cannot find module 'gap'. Did you mean to set the 'moduleResolution' option to 'node', or to add aliases to the 'paths' option?

203   import { GAP } from 'gap'
```
`"allowJs": false`にすると成功します。

@meganetaaan 
`bleserver.d.ts`のエラーについては、moddable側の型定義のexportを見直したら解消しますでしょうか。それとも特殊な事象があるでしょうか。

stc1988, 2024-09-19:
メモ：CIでFormatやLintがビルド後に行なっているのはビルド後に`tsconfig.json`が生成されるため。
本件の対応で`tsconfig.json`をgit管理対象とすることでビルド前にFormatやLintを行うことができ、CIで早期に検出可能となる

meganetaaan, 2024-09-19:
確認遅くなりました :pray:

>bleserver.d.tsのエラー

Moddable側の型定義のほうが新しいので、当該ファイルを削除してModdable側を使うことで解消しました。

>npm run generate-apidoc が失敗する

そもそもModdableの独自記法がそのままtscに渡されてSyntax Errorになります。tsconfig.jsonで除外設定を加えることでこちらも解消して、generate-apidocが通るようになりました。

差分を[こちらのブランチ](https://github.com/stack-chan/stack-chan/tree/stc1988-feature/moddable_typings)にpushしてあります。ご確認ください :ok_woman:

stc1988, 2024-09-19:
`@moddable/typings` を5.0.0に上げ、[Lint, Formatをビルド前に実行する](https://github.com/stc1988/stack-chan/actions/runs/10950459489/job/30405719506)こともできました。

差分を見直して今晩にでもPR出します

stc1988, 2024-09-21:
#289 がマージ済みのためクローズします

closed 2024-09-21 by stc1988