stack-chan/stack-chan #267: mods/face_tracker の接続先について

stc1988, 2024-08-23:
**既存の機能に問題がある場合、その説明**

手元で動作確認をしてみたいのですが、接続先のサーバーソフトなどはありますでしょうか。
キャプチャ画像や実装を見る限りは、顔認識には[M5Stack UnitV2](https://shop.m5stack.com/products/unitv2-ai-camera-gc2145?_pos=1&_sid=55e7e74e7&_ss=r)が必要でしょうか

stc1988, 2024-08-23:
あとmod自体も下記RT版の通りビルドが通らないので修正が必要です
https://github.com/rt-net/stack-chan/issues/14

meganetaaan, 2024-08-24:
はい。UnitV2のデフォルトアプリケーションです。
UnitV2のWeb UIまたはAPI経由で顔認識機能に切り替えた後でM5Stackから接続します。

https://github.com/rt-net/stack-chan/issues/14 の件もそうですね。一応RT版で修正する予定があったので聞いてみますね。
@KuraZuzu さんご無沙汰してます。 上記の対応状況はいかがですか？
直近rt-net/stack-chanで対応する予定がなければ、こちらで先に直しちゃおうかと思います。

KuraZuzu, 2024-08-26:
@meganetaaan さん。ご無沙汰しております。
https://github.com/rt-net/stack-chan/issues/14 ではFetchを使わずにtext/decoderをホスト側に入れて動作するよう、修正、動作確認中です。