stack-chan/stack-chan #272: ドキュメントのチェックツールを導入する

stc1988, 2024-08-25:
**既存の機能に問題がある場合、その説明**

#255 をはじめドキュメントの新規作成や修正することが多くなってきました。そこで、ソースコードにおける`prettier`や`ESLint`のようなツールを導入することにより、作業者間でのドキュメント体裁等の差異を無くしていきたい。

**あなたが望む解決策**

対応可能なものから段階的としたいですが、以下の対応を想定しています。

* 文章の表現方法や表記揺れの統一
  * `textlint`のようなツールを想定
  * 日本語・英語で適用ルールを手軽に分けることができるか
* markdownファイルのフォーマット統一
 * `prettier`で対応可能だが、他ツールと併用時して問題が生じないか確認が必要  
* markdownとしての体裁統一
 * `markdownlint` のようなツールを想定

他に改善できそうなアイデアや上記対応の懸念点などあればコメントお願いします

stc1988, 2024-09-07:
`textlint`と`prettier`を試してみて大きく気になった点は以下です

* 日本語ドキュメントにおけ英単語前後に空白文字を入れるか否か。
  * `textlint`では入れない。別途ruleセットで入れることも可能そう
  * `prettier`では入れる。`prettier`のバージョンを3系に上げると入れない挙動になる。markdownに対して細かい制御ができないのでフォーマットに無理してprettierを使う必要はないかもしれません。
* markdownにおける箇条書きをするときの記号をハイフンとするとかアスタリスクとするか。現状でも混在が発生している。

meganetaaan, 2024-09-19:
脱prettierしたので、フォーマットは他のツールを探す感じですね。
biomeもMarkdownの言語サポートが[ロードマップ](https://biomejs.dev/internals/language-support/)に入っているので、対応を待って導入する方針で良いと思います。

Linterは `moddable-jp` でも使っているtextlintが導入しやすそうです。[設定を真似しても](https://github.com/Moddable-OpenSource/moddable-jp/tree/dev/translate-jp/textlint)良いかなと。

stc1988, 2024-09-19:
そうですね、フォーマットとmarkdown文法のLintはbiomeの対応待ちでよさそうです。
文章の構成はtextlintを予定していますが、日本語(*_ja.md)と英語(*.md)ドキュメントで`textlintrc`を分けようか検討中です。
途中まで作業していたのですが、実装系の作業が終わったら再開予定です。

stc1988, 2025-02-02:
Biomeのmarkdown対応は、[2025年のロードマップとして対応予定](https://biomejs.dev/blog/)　とのこと。

stc1988, 2026-11-15:
@meganetaaan 
[Biome v2.6](https://biomejs.dev/blog/biome-v2-6/)からMarkdownやyamlが対応してます。
一方で、本issue起票時点とは異なり、現状ではほとんどAIエージェントによるドキュメント作成が主となっています。Formatter、Linterのような静的解析ベースでのドキュメント品質担保に加えて、文章のいわゆるAIっぽさを改善するような対応も必要です。

本課題はクローズして、別途issueを起票して検討するのはいかがでしょうか