stack-chan/stack-chan #286: ESList v8.xのEOL対応について

stc1988, 2024-09-14:
**既存の機能に問題がある場合、その説明**

ESLintが10月5日に[EOL](https://eslint.org/blog/2024/09/eslint-v8-eol-version-support/)を迎えます。
現状の開発に大きく支障をきたすものではありませんが、この期にバージョンアップをしたいです。

**あなたが望む解決策**。

ESLintおよびそれに関連するパッケージを最新にバージョンアップする。
合わせて[prettier](https://github.com/prettier/prettier)も古くなっているのでバージョンアップする。

**検討した代替案**

linterとformatterを兼ね備えた[Biome](https://biomejs.dev/ja/)へ移行してしまうのも管理パッケージが少くなるので選択肢として有りかと思います。

**その他**

ESLintとBiomeどちらを選んだ場合もTypeScriptも合わせたバージョンアップをしておきたいです。

meganetaaan, 2024-09-14:
起票ありがとうございます。ちょうどbiomeへの移行を考えていて、個人プロジェクトで試用していました。良さそうなので乗り換えたいと思います。
なるべくbiome標準のルールに従う形でコードも書き換えましょうか。

stc1988, 2024-09-14:
了解です。biome導入で進めましょう。
いったんbiomeでformatter, linterかけた状態でドラフトのPR作りますので、差分を見てルールの適宜カスタマイズを検討しましょう

stc1988, 2024-09-14:
PR作ると以降コミットごとにactionsが動いてしまうので、手元のブランチでformatまでかけました
https://github.com/stack-chan/stack-chan/compare/dev/v1.0...stc1988:stack-chan:feature/biome

以下が適用されたため全体的に差分が多い状況です。逆に下記以外の差異を見つけられない状況。
formatter任せとなれば個人的には従うまでですが、いかがでしょうか？　@meganetaaan 

* インデントがタブ
* クォートがダブルクォーテーション
* 命令の末尾にセミコロン

また、lintもかけてみましたが、手修正が必要なものが多く、対応内容が判断つきづらいルールは一時的に無効にして進めようかと思います

meganetaaan, 2024-09-14:
ごめんなさい、フォーマットについては前言撤回します。後ろの2つは良いにしてもインデントがタブだけはどうしても生理的に受け付けなくて…😭
挙げていただいた3つのルールは除外（現状に合わせたルールに変更）にしましょうか。

Lintについて了解しました。段階的に別PRで対応していき、指摘ゼロを目指す方向でいきたいですね。

stc1988, 2024-09-14:
元のprettierrcを参考にFormatルールを見直しましコミットしました。
差分は概ねフォーマット対象外だったjsonと末尾カンマ付与ぐらいになりました。

stc1988, 2024-09-14:
いくつか[lintのルールを無効化](https://github.com/stack-chan/stack-chan/commit/062a01c76faf1e198672927f9c496d3451e1f69a)にしてビルド通るところまで確認。

stc1988, 2024-09-19:
#287 でマージ済み。Lint推奨ルールの対応は #288 を別件起票しましたので、本件はクローズします

closed 2024-09-19 by stc1988