stack-chan/stack-chan #294: build flash -firmware

su-cheng-yang, 2024-09-22:
[image]

su-cheng-yang, 2024-09-22:
please help me

stc1988, 2024-09-23:
Hi, the link you posted is broken and I can't check the details. Could you please repost it?

su-cheng-yang, 2024-09-23:
Hi,Thanks for your reply，This issue is due to computer configuration problems，which have already been solved。I have encountered another problem：

npm run mod ./mods/face/manifest.json
### Error: no manifest!

please help me!

stc1988, 2024-09-23:
Please specify the `target` according to the device you are using, as shown below.

```
npm run mod --target=esp32/m5stack_core2 ./mods/face/manifest.json
```

su-cheng-yang, 2024-09-24:
I followed your instructions to run the command, but still received an error message. However, I changed -- target=esp32/m5stack_core2  to -- target=esp32, and now it can be run.

run as shown below:

s@ubuntu:~/stack-chan/firmware$ npm run mod --target=esp32/m5stack_core2 ./mods/face/manifest.json

> stack-chan@0.2.1 mod
> cross-env npm_config_target?=esp32/m5stack cross-env-shell mcrun -d -m -p $npm_config_target ./mods/face/manifest.json

### Error: No such file or directory
s@ubuntu:~/stack-chan/firmware$ npm run mod --target=esp32 ./mods/face/manifest.json

> stack-chan@0.2.1 mod
> cross-env npm_config_target?=esp32/m5stack cross-env-shell mcrun -d -m -p $npm_config_target ./mods/face/manifest.json

su-cheng-yang, 2024-09-24:
But I encountered another mistake，as shown below:

Python requirements are satisfied.
Please wait - probing for device
Failed to find a suitable device. Check your connections or set UPLOAD_PORT
Please wait - probing for device
Failed to find a suitable device. Check your connections or set UPLOAD_PORT
/bin/sh: 1: [[: not found
/home/s/Projects/moddable/tools/serial2xsbug/serial2xsbug_lin.c:132: No such file or directory
make: *** [/home/s/Projects/moddable/build/tmp/esp32/debug/face/makefile:136: debug] Error 1

However, I have already connected an m5stack.basic to the virtual machine using type_c

stc1988, 2024-09-24:
If you have recently set up the environment with `npm run setup`, it is likely due to an issue with Moddable. In that case, updating Moddable to the latest version 
should resolve the issue. 

https://github.com/Moddable-OpenSource/moddable/issues/1408

You can check the commit of the installed Moddable with `npm run doctor`.

su-cheng-yang, 2024-09-25:
I'm sorry，Following your advice,updating Moddable to the latest version, reproduced the same problem.as shown below:

Build environment:linux
Target device: esp32/m5stack
Steps to Reproduce:

Build and install the app using this build command: npm run mod --target=esp32/m5stack  ./mods/look_around/manifest.json
/bin/sh: 1: [[: not found
/home/s/Projects/moddable/tools/serial2xsbug/serial2xsbug_lin.c:132: No such file or directory
make: *** [/home/s/Projects/moddable/build/tmp/esp32/debug/look_around/makefile:133: debug] Error 1

stc1988, 2024-09-25:
It's true that there is a failure, but from the logs, it seems that another issue is occurring.

How did you update Moddable?

su-cheng-yang, 2024-09-25:
I will perform the following actions：

git clone https://github.com/Moddable-OpenSource/moddable

su-cheng-yang, 2024-09-27:
I made another attempt，Stuck at the following prompt：

s@ubuntu:~/stack-chan/firmware$ npm run mod --target=esp32/m5stack  ./mods/face/manifest.json
> stack-chan@0.2.1 mod
> cross-env npm_config_target?=esp32/m5stack cross-env-shell mcrun -d -m -p $npm_config_target ./mods/face/manifest.json

Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Constraint file: /home/s/.espressif/espidf.constraints.v5.3.txt
Requirement files:
 - /home/s/esp32/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/s/.espressif/python_env/idf5.3_py3.8_env/bin/python
Python requirements are satisfied.
/bin/sh: 1: [[: not found
       
    
Has this open-source project already rejected interested developers  ，please  reply  me ,thank you .

stc1988, 2024-09-27:
As you continue developing stack-chan, I recommend understanding Moddable as well.

For this Moddable update, simply updating the repository is not sufficient. Please refer to the following documentation for the updating and other troubleshooting.

https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md

su-cheng-yang, 2024-09-27:
thank you for your reply

closed 2024-09-27 by su-cheng-yang