stack-chan/stack-chan #307: M5Stack CoreS3でPSRAMを有効にすると極端に動作が遅くなる

meganetaaan, 2024-12-19:
**不具合の概要**
M5Stack CoreS3でPSRAMを有効にすると極端に動作が遅くなります。

**再現手順**
1. M5Stack CoreS3ターゲットでビルドする
2. 動作が重くなる
  - 瞬きなどのアニメーションが遅延する
  - シリアルサーボと通信できなくなる

**環境 (次の項目を埋めてください):**

- Moddable v5.3.0
  - ESP-IDF v5を使うバージョンで発生している
- 使用するM5Stackの種類

meganetaaan, 2024-12-19:
アールティ版によるとPSRAMをOFFにすると動作するとのこと。

https://github.com/rt-net/stack-chan/blame/a6f237005841d12ceadfca7f33080e05a754b94b/firmware/docs/flashing-firmware_ja.md#L43

```
前述の通りｽﾀｯｸﾁｬﾝのファームウェアは基本プログラム（ホスト）とユーザアプリケーション（MOD）から構成されます。  
esp-idfのバージョンが5.xからesp32に接続しているPSRAMの扱いが変わっています。デフォルト設定ではスムーズに動かないため一部変更します。  
~/.local/share/moddable/build/devices/esp32/targets/m5stack_cores3/sdkconifgのファイルにある
CONFIG_SPIRAM=yをCONFIG_SPIRAM_nに書き換えます。
```

meganetaaan, 2024-12-19:
アールティ版ではModdable側のsdkconfigを変更していますが、 `stack-chan` 側でsdkconfigを上書きできないか確認してみます

stc1988, 2024-12-19:
機種ごとにsdkconfigファイルを分ける方針で良いかと思います🙆

stc1988, 2025-06-15:
何らか対策が入りそうですが、現時点ではアプリケーション側からの設定ができないようです

https://github.com/Moddable-OpenSource/moddable/discussions/1502

meganetaaan, 2025-11-20:
サーボドライバ側の非効率なコードを見直すことで、Moddable側に手を加えずにPSRAMを有効化した状態で動作するようになりました。
https://github.com/stack-chan/stack-chan/pull/352