stack-chan/stack-chan #312: Build Failed with Moddable 5.4.0

meganetaaan, 2025-01-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Moddable v5.4.0
2. `npm run build`
3. See  errors bellow

```
ble/beacon-packet.ts:54:22 - error TS2693: 'Bytes' only refers to a type, but is being used as a value here.

54     const uuid = new Bytes(payload.slice(2, 18).buffer, true)
                        ~~~~~

ble/beacon-packet.ts:72:16 - error TS2693: 'Bytes' only refers to a type, but is being used as a value here.

72     return new Bytes(this.#payload.slice(2, 18).buffer)
                  ~~~~~
```

stc1988, 2025-01-26:
すみません、　ModdableSDKの[#1452](https://github.com/Moddable-OpenSource/moddable/pull/1452) での対応漏れによるものです。[#1455](https://github.com/Moddable-OpenSource/moddable/pull/1455 で対応をしているので次回ModdableSDK更新に取り込まれる予定です

stc1988, 2025-01-31:
5.4.1がリリースされたので、本リポジトリの定義を取り除いてビルドできるか確認してPR出します

meganetaaan, 2025-02-02:
アップストリームの型定義の干渉が解消されてビルドが通るようになりました。こちらはクローズします🙆

closed 2025-02-02 by meganetaaan