stack-chan/stack-chan #316: chatAudioIO を組み込む

stc1988, 2025-04-06:
**既存の機能に問題がある場合、その説明**

現在の音声対話機能はSTT-> LLM -> TTSといくつかのサービスを組み合わせて実現しているため応答性がそこまで高くない。
一方で、最近のModdableSDKの更新では、リアルタイム会話モジュールとして [chatAudioIO](https://github.com/Moddable-OpenSource/moddable/tree/5.6.0/modules/network/services/chatAudioIO)が追加されました。これをstack-chanでも利用できるようにしたい。

**あなたが望む解決策**。

* chatAudioIOはいくつかのネイティブコードを含むため、ホスト側でmanifestファイルをインクルードする必要があります。
* microphoneクラスが、`pins/audioin`を使用としており、chatAudioIOが使用する`embedded:io/audio/in` と競合するため、microphoneを`embedded:io/audio/in`ベースへの改修が必要

**その他**

解決策を実施することで少なくともmodからはchatAudioIOを利用できるようなりそうですが、stack-chanホスト側のインターフェース(robotクラス)もリアルタイム音声対話への対応を検討してもよいかと考えています。この話題自体はDiscussionとかに切り出してもよいかも。

meganetaaan, 2025-05-19:
robotをリアルタイム音声対話に対応させる旨を含めて、賛成です。
Ecma419対応がこの1年で相当進んだので、ｽﾀｯｸﾁｬﾝも全面的に `pins` から `embedded:io` に移行したいですね。

stc1988, 2025-05-19:
`ChatAudioIO` の[型定義](https://github.com/Moddable-OpenSource/moddable/blob/public/typings/ChatAudioIO.d.ts)が追加されたため、ホストにもすぐ組み込めそう。

あと気になったのは、OpenAIなどのAPIキーがconfigからの参照でしか設定できないので、既存のAI機能の設定との統合方法の検討が必要

meganetaaan, 2025-11-20:
>会話が続くと例外発生しますが

別件のissue（CoreS3+PRRAM有効化時に落ちる）の調査時に、以下のコンボでプログラムが落ちる事象を観測しました。

* 瞬きなどの顔描画更新時に全体の動作が重くなる
* サーボドライバ内のTimer.repeatは厳密に、指定時間おきに繰り返し実行される
* Timerのコールバックの中ではPromiseオブジェクトを作成するが、動作が重いためにPromiseの解決前に次のタイマーコールバックが実行されてメモリの使用が増え続ける
* メモリ不足で落ちる

[サーボドライバのリファクタ](https://github.com/stack-chan/stack-chan/pull/352)を実施して上記事象が解消されたので、ChatAudioIOを動かしても落ちなくなったかもしれません。
（Timer#repeatを使わずに処理の完了時に、次回のタイマー1回分を予約するようにして、タイマーが重複しないようにしました）