stack-chan/stack-chan #319: stack-chan/firmware/stackchan/sdkconfig をディレクトリとして期待している処理とファイルとして期待している処理がある

kanekoh, 2025-05-11:
**不具合の概要**
ビルド時に以下のエラーが出力される。npm run build コマンド自体は異常終了しないので影響ないですが気になります。

IsADirectoryError: [Errno 21] Is a directory: '/home/runner/work/stack-chan/stack-chan/firmware/stackchan/sdkconfig'

**再現手順**
不具合を再現できる手順を記述してください
1. 以下のコマンドを実行する
    npm run build --target=esp32/m5stack_cores3

**想定する挙動**
想定する挙動を記述してください。

エラーが出力されない状態

**ログ**
Python being checked: /home/runner/.espressif/python_env/idf5.3_py3.12_env/bin/python
Traceback (most recent call last):
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf.py", line 844, in <module>
    main()
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf.py", line 7[25](https://github.com/kanekoh/stack-chan/actions/runs/14941842923/job/41980079807#step:6:26), in main
    cli = init_cli(verbose_output=checks_output)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf.py", line 687, in init_cli
    all_actions = merge_action_lists(all_actions, extension.action_extensions(all_actions, project_dir))
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/.local/share/esp[32](https://github.com/kanekoh/stack-chan/actions/runs/14941842923/job/41980079807#step:6:33)/esp-idf/tools/idf_py_actions/dfu_ext.py", line 70, in action_extensions
    return dfu_actions if is_target_supported(project_path, SUPPORTED_TARGETS) else {}
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf_py_actions/tools.py", line 721, in is_target_supported
    return get_target(project_path) in supported_targets
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf_py_actions/tools.py", line 103, in get_target
    return get_sdkconfig_value(path, 'CONFIG_IDF_TARGET')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/.local/share/esp32/esp-idf/tools/idf_py_actions/tools.py", line 709, in get_sdkconfig_value
    with open(sdkconfig_file, 'r') as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^
IsADirectoryError: [Errno 21] Is a directory: '/home/runner/work/stack-chan/stack-chan/firmware/stackchan/sdkconfig'
```log
```

**スクリーンショット**
エラーログのみ添付しました。

**環境 (次の項目を埋めてください):**

- OS: [Windows]
- IDE: [VSCode]:
- ｽﾀｯｸﾁｬﾝのバージョン: github の最新
- 使用するM5Stackの種類 M5StackCoreS3

**その他**

私が確認したのは以下では、sdkconfig がディレクトリとして存在することを期待していますが、エラーを出力したesp-idf はファイルを期待しているようでした。
```
    "esp32/m5stack_cores3": {
      "build": {
        "PARTITIONS_FILE": "./sdkconfig/customPartitions.csv"
      },

```

stc1988, 2025-05-11:
手元では再現しなかったのでModdableないしESP-IDFのバージョン問題の可能性があります。
npm run doctor の結果を貼っていくいただけますでしょうか

kanekoh, 2025-05-13:
再確認したところ ESP-IDF v5.3.1 の際には出力されていたようですが、現在 v5.4 を利用しており、試したところエラーが出力されないことを確認しました。
お騒がせしました。

現在の環境の doctor の結果です。再現しないことが確認できているため、不要な情報かもしれませんが添付します。
```
$ npm run doctor

> stack-chan@0.2.1 doctor
> echo stack-chan environment info: && git rev-parse HEAD && git rev-parse --show-toplevel && xs-dev doctor

stack-chan environment info:
a6b1c61a46865631b79b8862ccb7410d051a280f
/home/hiroyuki/workspace/stack-chan
xs-dev environment info:
  CLI Version                0.37.0
  OS                         Linux
  Arch                       x64
  Shell                      /bin/bash
  NodeJS Version             v22.15.0 (/home/hiroyuki/.nvm/versions/node/v22.15.0/bin/node)
  Python Version             3.12.3 (/usr/bin/python)
  Moddable SDK Version       5.6.0 (/home/hiroyuki/.local/share/moddable)
  Supported target devices   linux, esp32
  ESP32 IDF Directory        /home/hiroyuki/.local/share/esp32/esp-idf

```

meganetaaan, 2025-05-19:
xs-dev v0.37.0 + ESP-IDF 5.4.0でビルドができていますね。
リポジトリで指定するxs-devのバージョンも https://github.com/stack-chan/stack-chan/pull/318 でv0.37.0にするよう対応いただいています（ありがとうございます！）

closed 2025-05-19 by meganetaaan